openbikesensor/OpenBikeSensorFirmware #1: writing issues

mega-stoffel, 2020-05-03:
to avoid writing issues to the SD card, it might be good to wait until the bike (including the device) is not moving.
Without the shocks from the potentially rough roads, it might be more secure to execute write access to the SD card.

tobst, 2020-05-03:
I was assuming that power off while writing could be a problem. A quick fix is to reduce the the number of write accesses and ask the user to push the button while powering off. But looking into th effects of rough roads is agood idea. The code does not handle failed writing attempts at all so far.

4n08o, 2020-07-11:
This would need an accel sensor first. -> HW change.
Improving the HW by PCB (removing all the hand wired cables) should improve the behaviour already.

closed 2020-09-09 by Natenom