openbikesensor/OpenBikeSensorFirmware #7: Include metadata in output file

tobst, 2020-05-03:
What was the user configuration, how long did it take for a gps fix, etc

4n08o, 2020-07-11:
Sensor identifier, if not in the file name (see #6)

amandel, 2020-10-24:
There was also a _discussion_ in Slack triggered by @tobst about the metadata (https://openbikesensor.slack.com/archives/CRTNL8RQF/p1602699726018600 / https://openbikesensor.slack.com/archives/CRTNL8RQF/p1602702652024100). @mega-stoffel  brought the need for different settings for the OBS in (switching bike / might be sharing a sensor).

I try to collect some data here - feel free to edit the post and bring in your opinion. 

Eventually we might implement preset support where you can select one preset when the OBS is started. So I see the following places where we could store data:

### Storage Location

Name|Description
-|-
`CSV` | Most frequent as data created on the OBS and stored in a CSV record
`CSV-Header` | Stored as header / meta data upon creation of the CSV file
`OBS-Preset` | Data that is available as configuration data in the OBS but might be different in future per preset.
`OBS` | Data stored as configuration in OBS as part of the global configuration.
`OBS-Static` | Data that is stored in the OBS fix in the firmware
`Web-Portal` | Data that is stored (and managed) on the web portal account

### Mandatory

Is the value mandatory or optional? Might be a different discussion.

### Critical personal data

TODO: Flag data that needs to be kept secret / be protected / deleted after a while or on request

### Data Types

Name|Description
-|-
`OBS-Firmware-Version` | Version string of the OBS firmware, must also allow to distinguish self built versions
`OBS-Config`| OBS internal configuration data like display settings (we might list this elsewhere?)
`OBS-Server-Config`| OBS _server_ configuration of the wifi to be used and other stuff for OBS server mode
`Privacy-Areas`| Areas that are out of bound for data collection
`OBS-User-ID`| Id to identify and authenticate a user on the web portal.
`Bike-Type` | Enum? (TODO)
`E-Bike`|Enum? (no/Pedelec/speed Pedelec/ ??)
`Offset-Left` `Offset-Right`| 2x 0-999cm
`GPS-Quality`| Number of sats (last received timestamp??)
`Helmet`| (yes/no/sometimes) Driving with helmet?
`Measurement Date-Time` |  
`Measurement Position` | latitude / longitude / course / speed
`Measurement Distance` | 2 x 0-999cm
`Measurement Confirmed` | yes/no
`Measurement inside privacy area` | yes/no
`EMail` | User contact email address
`Avatar` | Link to user image

### Matrix

_Data Type_ | `CSV` | `CSV-Header` | `OBS-Preset` | `OBS` | `OBS-Static` | `Web-Portal`
----------- | -------|---------------|-|-|-|-
`OBS-Firmware-Version` | | :grey_exclamation: | | | :grey_exclamation: | 
`OBS-Config` | | | :grey_exclamation: |  | |  
`OBS-Server-Config` | | | | :grey_exclamation: | 
`Privacy-Areas` | | | :grey_exclamation: |  | |  
`OBS-User-ID` | | :grey_exclamation: (login)| :grey_exclamation: |  | |  
`Bike-Type` | | :question: | :question: |  | |  
`E-Bike` | | :question: | :question: |  | |  
`Offset-Left` `Offset-Right` | | :grey_exclamation: | :grey_exclamation: |  | |  
`GPS-Quality` | :grey_exclamation: | 
`Helmet` | | :question: | :question: |  | |  
`Measurement Date-Time` | :grey_exclamation: | | 
`Measurement Position` | :grey_exclamation: | | 
`Measurement Distance` | :grey_exclamation: | | 
`Measurement Confirmed` | :grey_exclamation: | | 
`Measurement inside privacy area` | :grey_exclamation: | | 
`EMail` | | | | | | :question:  
`Avatar` | | | | | | :question: 

### Symbols:

Mandatory: :grey_exclamation:
Optional: :question:
No way: :no_entry_sign:

mega-stoffel, 2020-11-10:
I think, it could be helpful to have the information about privacy-zones in the uploaded csv file.
The csv file should have the information, if privacy-zones were configured during the recording and how the according option was set.
If no privacy-zone was configured and/or the option "ignore privacy zone" is set, there should be a warning/information in the HLRS portal, at the latest, when pressing on "visible/publish track".

amandel, 2021-01-13:
Actualy we have the data in the CSV file but not the other -> Re-opening