openbikesensor/OpenBikeSensorFirmware #12: Add a Fritzing project to document the exact wiring

boldt, 2020-05-06:


tobst, 2020-05-08:
Last time I looked into Fritzing, I could not find all the hardware components used in this project.

tobst, 2020-05-23:
There are parts for the ESP32 and the HC-SR04(P) sensor. 
The display can be found here: https://forum.fritzing.org/t/oled-128x64-i2c-monochrome-display-ssd1306-created/1202/4
This GPS could be used, but ours has no PPS-pin: https://forum.fritzing.org/t/u-blox-neo-6m-gps-breakout/6107
SD-Card: http://fritzing.org/projects/microsd-sd-raspberry-pi-adapter
I could find non of the charging/battery protection modules

Halbelunge, 2020-06-04:
Maybe I can take the schematic from the pcb development, remove everything that is additional and post this schematic? Will not be a direkt wiring but it shows all connections. Would take ~30min to do that.