openbikesensor/OpenBikeSensorFirmware #20: Check and enhance display during measurement

tobst, 2020-05-14:
The font size selected seemed to be the biggest fitting on the screen. In order to add status information like number of satellites etc. position and size should be adapted and a suitable font size for the status line as well as display class member functions for displaying should be added.

uwewoessner, 2020-05-18:
I had to use a smaller font anyway because the generated font header was broken.
We probably could use a larger font again if we limit it to numbers only.
The font is still too large to display both distances

boldt, 2020-05-22:
Fixed with b0aea8212ec62cbddccc23d6cc5f9305b586fba9

[image]

closed 2020-05-22 by boldt