openbikesensor/OpenBikeSensorFirmware #24: Switch to PlatformIO

uwewoessner, 2020-05-16:
Hallo alle miteinander,
ich würde das Projekt gerne von der Arduino IDE weg auf PlatformIO umstellen, was haltet Ihr davon?
Vorteile von PlatformIO:
Man kann gleichzeitig unterschiedliche Projekte bearbeiten die unterschiedliche library dependencies haben oder unterschiedliche hardware Platformen nutzen
VS Code macht viel besseres Syntax Highlighting und Auto Completion
OTA funktioniert zuverlässiger.
Der Compiler ist besser, er hat z.B. auch das Font Problem bemängelt (siehe nächsten commit)
Git integration.
Nachteile:
Ich hab jetzt alle meine eigenen Projekte umgestellt, sowohl esp32 als auch 8266, mir ist noch kein Nachteil aufgefallen.
Ein Problem bei der Umstellung gibt es jedoch, soviel ich weiss gibt es keine Möglichkeit beide IDEs gleichzeitig zu unterstützen, da PlatformIO source files mit .cpp will und Arduino mit .ino
Unter linux könnte man das evtl. mit Links machen aber geht wahrscheinlich unter Windows wieder nicht.

Was haltet ihr davon?

boldt, 2020-05-18:
Ich habe leider keine Erfahrung mit PlatformIO.

boldt, 2020-05-18:
Ich teste PlatformIO gerne mal. Wir haben ja das folgende Board:

https://www.az-delivery.de/collections/bestseller/products/esp32-developmentboard

In der Beschreibung steht:

> Der AZDelivery ESP32, welcher zu 100 % baugleich mit dem Original des Herstellers Espressif ist

Welches Board hast Du in PlatformIO gewählt? Ich finde:

* AZ-Delivery ESP32 Dev Kit C V4
* Espressiv ESP32 Dev Modul 

Auf der Webseite steht nichts von "C V4".

boldt, 2020-05-18:
Ich habe nun
* Das `Espressiv ESP32` gewählt
* libs dazu gepackt
* erfolgreich gebaut

PlatfromIO überzeugt !

boldt, 2020-05-18:
Wir sollten uns in jedem Fall auf eine IDE einigen. Meine Argumente für Platform IO:

* VS Code ist als Profi-IDE viel besser als die Arduino Anfänger-IDE
* Syntax Highlighting (insb. Fehler)
* Auto Completion
* Man kann damit #25 lösen
* Dependencies sollten damit auch immer bei allen gleich sein

tobst, 2020-05-18:
Also dann PlatformIO!

4n08o, 2020-07-11:
This issue can be closed!
Follow up with #78.

closed 2020-07-11 by boldt