openbikesensor/OpenBikeSensorFirmware #25: Line endings

boldt, 2020-05-18:
We should define which line ending we want to use in the whole project, since WIndows, Kinux and MAC use diffenrt one.

@uwewoessner - see this commit:

* https://github.com/Friends-of-OpenBikeSensor/OpenBikeSensorFirmware/commit/171118b1d367f5b85606b750ca895f658b5206d7

Unfortunately, the whole file changed - but I guess you just cahnged some lines.

boldt, 2020-05-18:
If I change one line, I have the same issue now:

[image]

boldt, 2020-05-18:
@uwewoessner - you changed the files to Windows file endings. 

TODO:

* [ ] Convert all file to the same file ending
* [ ] Add a `.gitattribues`, which handles the line endings for all users

boldt, 2020-05-18:
Could be fixed with #24

boldt, 2020-05-18:
# Analysis

## Windows Endings (CRLF) 

```bash
dennis@denpc:~/git/OpenBikeSensorFirmware$ file *.h *.ino | grep CRLF
config.h:                   C source, ASCII text, with CRLF line terminators
configServer.h:             C source, ASCII text, with CRLF line terminators
displays.h:                 C++ source, ASCII text, with CRLF line terminators
sensor.h:                   C++ source, ASCII text, with CRLF line terminators
config.ino:                 ASCII text, with CRLF line terminators
configServer.ino:           HTML document, UTF-8 Unicode text, with CRLF line terminators
OpenBikeSensorFirmware.ino: C source, ASCII text, with CRLF line terminators
sensor.ino:                 ASCII text, with CRLF line terminators
```

## Linux Endings

```bash
dennis@denpc:~/git/OpenBikeSensorFirmware$ file *.h *.ino | grep -v CRLF
font.h:                     ASCII text, with very long lines
gps.h:                      C source, ASCII text
logo.h:                     ASCII text
vector.h:                   C++ source, ASCII text
writer.h:                   C++ source, ASCII text
gps.ino:                    ASCII text
ota.ino:                    ASCII text
writer.ino:                 ASCII text
```

See: https://stackoverflow.com/a/3570574/605890

uwewoessner, 2020-05-18:
added  .gitattributes, should be fixed now

closed 2020-05-18 by uwewoessner