openbikesensor/OpenBikeSensorFirmware #26: WIFI crashed quite often

boldt, 2020-05-19:
On my ESP32, the WIFI crashes a lot. An example:

```
...........Initalize SoftAP AP successful.
mDNS responder started
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400debf8: 0020c000 20c0b138 27072c40
Core 1 register dump:
PC      : 0x400debfe  PS      : 0x00060a30  A0      : 0x800dec74  A1      : 0x3ffb1da0  
A2      : 0x3ffc12e8  A3      : 0xcde27b3e  A4      : 0x00000001  A5      : 0x3ffd26d8  
A6      : 0x3ffb1e0c  A7      : 0x3ffc23e8  A8      : 0x800debf9  A9      : 0x3ffb1d80  
A10     : 0x3ffb1dac  A11     : 0x3ffd26d8  A12     : 0x3ffc23e8  A13     : 0x3ffd26ec  
A14     : 0x3ffd26fc  A15     : 0x3ffd26d8  SAR     : 0x00000010  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400debfe:0x3ffb1da0 0x400dec71:0x3ffb1df0 0x400d6069:0x3ffb1e40 0x400d659c:0x3ffb1eb0 0x400e2eab:0x3ffb1fb0 0x4008954d:0x3ffb1fd0
```

The Exceltion decoder in the Arduino IDE says:

```
Decoding stack results
0x400debfe: WebServer::on(String const&, HTTPMethod, std::function , std::function ) at /home/dennis/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WebServer/src/WebServer.cpp line 258
0x400dec71: WebServer::on(String const&, HTTPMethod, std::function ) at /home/dennis/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WebServer/src/WebServer.cpp line 253
0x400d6069: startServer() at /home/dennis/git/OpenBikeSensorFirmware/configServer.ino line 241
0x400d659c: setup() at /home/dennis/git/OpenBikeSensorFirmware/OpenBikeSensorFirmware.ino line 181
0x400e2eab: loopTask(void*) at /home/dennis/.arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/main.cpp line 14
0x4008954d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

This seems to be an issue with the esp32 firmware.

tobst, 2020-05-19:
Could also be a hardware issue. The other devices have the az-delivery esp32 and i haven't heard of wifi issues. But I have similar and other errors trying the same esp32 you have.

boldt, 2020-05-20:
Didnt show up anymore. Probably a "defective contact" on the breadbaord.

boldt, 2020-05-21:
* The Issue happend, when the ESP32 is connected via USB only
* The issued does not happen, when the ESP32 with additional energy

closed 2020-09-16 by boldt