openbikesensor/OpenBikeSensorFirmware #27: Screen content displaced

Natenom, 2020-05-19:
Sometimes the bottom part of the measured value is on top of the screen and the top part is on the bottom. Must turn off/on to fix that.

boldt, 2020-05-20:
I did plenty of improvements on the display (no new release yet). 
Can you show me a picture of this issue to understand it fully? Thx.

Natenom, 2020-05-21:
https://files.natenom.com/s/1719d.jpg

boldt, 2020-05-22:
Ich glaube das tritt auf, wenn die Batterie sich dem Ende neigt. Dann ist das Display unterversorgt und  "buggt" - zumindest scheint das bei dem OBS von @mega-stoffel der Fall zu sein. Ich beobachte weiter

boldt, 2020-05-22:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

boldt, 2020-09-16:
Hardware issue, which is fixed with a pull-up resistor on the board

closed 2020-09-16 by boldt