openbikesensor/OpenBikeSensorFirmware #51: Maximum program size

boldt, 2020-05-23:
> Sketch uses 912706 bytes (69%) of program storage space. Maximum is 1310720 bytes.

* 1310720 / 1024 = 1280 kByte = 1,25 MB

What causes memory:

* If you do not need OTA firmware update abilities, switch to a single partition
* Remove the SPIFFS partition

In Arduino IDE, we can use `partition scheme: Minimal SPIFFS (1,9MB APP with OTA)`

See: 
* https://esp32.com/viewtopic.php?f=19&t=14420
* https://forum.arduino.cc/index.php?topic=578847.0

boldt, 2020-05-27:
https://sparks.gogo.co.nz/avr-ram-use.html

4n08o, 2020-07-11:
I presume, that our binary gets bigger and bigger.  We have huge memory on the SD Card.  Just for the update, is it feasible to temporarily store the OTA binary there and update from there?

amandel, 2020-11-10:
With v0.3. - merging the bluetooth branch the new partition schema was rolled out to the master.

closed 2020-11-10 by amandel