openbikesensor/OpenBikeSensorFirmware #52: Configuration with the button w/o WLAN

boldt, 2020-05-23:
> Idee: Meine Actioncam hat einen sogenannte Housing Mode. Da kann man alles einstellen mit nur einem Knopf. Kurz drücken wechselt im Menü eins weiter. Lange drücken ändert etwas. Könnte man das hier auch so machen? Dann bräuchte man kein wlan zum konfiguroeren. Wäre nur verfügbar, wenn man wlan mode oder dann halt im devel mode startet.

4n08o, 2020-07-11:
I would suggest at least a second button, considering the options/configurations we already have in addition to the functionality of the button during normal operation (we may need the long press for something...).
HW Change

amandel, 2021-02-07:
I did implement a way to upload the collected track data in server mode without the need to connect a browser. What other frequently used use cases would be beneficial?