openbikesensor/OpenBikeSensorFirmware #60: Doppeldruckfunktion fuer kaputte Werte

boldt, 2020-05-25:
> Motorrad mit knatterndem Motor überholt mit 3 Metern abstand; Gerät zeigt 20 cm an. Ahnlich bei Lkw oder lauten Autos. Problem ist bekannt. Ich möchte die Doppeldruckfunktion haben, um zwar einen Überholvorgang bestätigen zu können, aber auch in den Daten zu hinterlegen, dass die Messung nicht valide gewesen sein kann.

> Die Doppeldruckfunktion sollte den "Confirmed"-Wert auf z. B. 2 setzen statt ein;damit klar ist, dass zwar überholt wurde, der Wert aber unrealisitsch war.

amandel, 2020-10-14:
Doppeldruck bedeutet immer, dass die Reaktion auf den Einzeldruck erst verzögert erfolgen kann (es könnte ja noch der 2. Druck kommen, das muss abgewartet werden). Was wäre mit länger Drücken, so eine Sekunde und dann auch Rückmeldung im Display?
Ich würde lieber gar kein Knopf drücken müssen (klar, das geht nicht) wie werden die unbestätigten Daten Ausgewertet? Ich bin noch nicht viel mit OBS gefahren, aber es gibt ja reichlich Messungen die kein Überholvorgang sind, Autos im Gegenverkehr, überholende Radfahrer, enge Straßen, ... sollte die dann auch immer weg drücken?

amandel, 2021-01-02:
If there are no objections I'd close this one.

closed 2021-02-07 by amandel