openbikesensor/OpenBikeSensorFirmware #62: Detect empty files

boldt, 2020-05-26:
On boot, detect empty files and use them

> Viele Dateien sind 63 Byte groß (keine Daten, nur Header von CSV)

amandel, 2020-12-07:
We also have the demand to rename the file once we have a valid time -> `2020-17-07T16.00.00-AAAA.obstrack.csv` <- I have to think if we calculate the time back to the beginning of the track, AAAA would be the device id. Related to #95 and #113 

There is no need to write the file when we create the header, we could just fill the buffer and so never create files that only contain header data.

mega-stoffel, 2020-12-08:
Please don't use something like 2020-17-07, that's not useful at all. I'd prefer (in this case) 2020-07-17.

amandel, 2020-12-08:
Sure

commit 1d341ee0 (2020-12-12): Change filename once we know the time (#173)

* Change filename once we know the time (fixes #95)
* Use a different name if the file is in upload already (fixes #113)
* Do not write file already with header creation to avoid empty files (fixes #62)
closed 2020-12-12 by amandel