openbikesensor/OpenBikeSensorFirmware #69: Hardware parts links

tobst, 2020-06-06:
Some links are unavailable after a while. For some parts exact measurements are important however, so just providing a shopping-link can lead to purchasing a similar part with wrong dimensions which will not fit into the case.

boldt, 2020-06-07:
How about moving everything related to the hardware into the Meta repo?
Or create an additional "Hardware" repo?

Halbelunge, 2020-06-14:
I think providing product photos, measurments and search keywords in an extra document in one of the repos (or both) would be the best.
Even if one link is dead, it would be easier to find a similar product yourself.

amandel, 2020-12-26:
Was this moved to the documentation area already? Can it be closed?