openbikesensor/OpenBikeSensorFirmware #72: Handle button in a separate class

boldt, 2020-06-07:
Maybe there are libs?

boldt, 2020-06-07:
Discussions:

* https://forum.arduino.cc/index.php?topic=14479.0
* https://forum.arduino.cc/index.php?topic=425587.0
* https://stackoverflow.com/questions/32811098/arduino-click-double-click-and-hold-button
    * Debounce !!
    * https://www.arduino.cc/en/Tutorial/Debounce

Libs:

* https://github.com/poelstra/arduino-multi-button
* http://www.mathertel.de/Arduino/OneButtonLibrary.aspx
    * Works with ticks
* https://playground.arduino.cc/Code/Button/

boldt, 2020-06-07:
# Buttons and interrupts

* https://www.arduino.cc/reference/de/language/functions/external-interrupts/attachinterrupt/
* https://www.mikrocontroller.net/articles/Verwendung_von_Interrupts_mit_Arduino
* http://diwo.bq.com/de/interrupts-mit-arduino-benutzen/
* https://lastminuteengineers.com/handling-esp32-gpio-interrupts-tutorial/#attaching-interrupt-to-a-gpio-pin

# Libs with Interrupts

* https://github.com/Alarm-Siren/arduino-buttons
* https://github.com/blackketter/Switch
* https://github.com/t3db0t/Button
* https://www.instructables.com/id/Arduino-Push-Switch-Debouncing-Interrupts/

Halbelunge, 2020-06-14:
Die Lib hier könnte auch passen
[https://github.com/mathertel/OneButton]

4n08o, 2020-07-11:
Concept for a button press strategy first?
short press: confirm measurement
long press: enter config?
short double press: cancel measurement (#60)
...?

amandel, 2020-12-26:
See also #136

PR #268 (2021-08-04): Debounce button read
also fixes #72
commit 98ffc3a8 (2021-08-04): Debounce button read (#268)

* Debounce button read
* also fixes #72
closed 2021-08-04 by amandel