openbikesensor/OpenBikeSensorFirmware #75: GPS-Status aufnehmen

mega-stoffel, 2020-06-18:
auf der Seite, wo man die aktuelle GPS-Position als Privacy-Zone übernehmen kann, wird auf einen GPS-Fix gewartet. Allerdings dauert das eeeeeewig. Da nicht klar ist, ob der Fix tatsächlich nicht da ist, oder ob nur nicht "weitergemacht" wird, sollte der GPS-Status auf diese Seite aufgenommen werden. Ähnlich wie beim Display, z.B. die Anzahl der Satelliten anzeigen und - bestenfalls - dynamisch anpassen; oder mit einem weiteren Knopf "Re-Check GPS Status".

boldt, 2020-06-20:
Ggf. sollten wir hier einfach auch die GPS-Position des Browsers verwenden, wenn man sich mit einem Hand verbindet.

4n08o, 2020-07-11:
Nur wenn GPS-fix da ist, Option Privacy ueberhaupt im Display anbieten sonst ausgrauen oder auf Browser verweisen?

boldt, 2020-07-11:
Hier sollte man eher regelmäßig mit AJAX such die aktuelle GPS-Position vom Sensor holen, ohne dass man die Seite neu laden muss.

Außerdem sollte die GPS-Position auch manuell eingeben können. Daher ist ausgrauen nicht sinnvoll.

amandel, 2021-02-25:
See also https://openbikesensor.slack.com/archives/C012H97P10E/p1614247128008000

commit a54e5235 (2021-08-04): Renew privacy area handling in config server (#80) (#266)

- separate GPS status display and `Make Current Location Private` function. fixes #80
- allow to edit all available privacy areas on the page, fixes #107
- redirect to the privacy area page after update. fixes #77
- pull current gps status on page, fixes #75
- fixes #74
- redirect to main page if privacy area data was not changed
closed 2021-08-04 by amandel