openbikesensor/OpenBikeSensorFirmware #88: Rename folder `OpenBikeSensorFirmware` to `src`

boldt, 2020-07-06:
Wha do you hink @tharkay @DanielHabenicht 

(**Hint**: Use `git mv`)

tharkay, 2020-07-08:
The Arduino IDE needs the folder name and the .ino file name to match (for whatever reason). I totally agree, that a `src` directory would look much better! After changing the folder name, I get this error:  [image]

Changing this folder would make the project incompatible with the Arduino IDE. Although the question remains if we still want to support that IDE ...

amandel, 2020-11-15:
Any objections against renaming? Is someone still using Arduino IDE for OBS development - please shout ;)

commit b6c8ed9e (2020-11-29): Rename source folder `OpenBikeSensorFirmware` to `src`

Fixes #88
closed 2020-11-29 by amandel