openbikesensor/OpenBikeSensorFirmware #89: Sources for external libs not mentioned

Halbelunge, 2020-07-26:
The external libs, like SSD1306.h are not mentioned to install. So there is no source for these mentioned (there are multiple ones as an option).

Maybe it is better, to include them into the project instead of rely on external git repos.

boldt, 2020-07-27:
* [The libs are described here](https://github.com/Friends-of-OpenBikeSensor/Friends-of-OpenBikeSensor.github.io/blob/master/software/firmware/setup.md)
* [We use platformIO, which handles the dependencies](https://github.com/Friends-of-OpenBikeSensor/OpenBikeSensorFirmware/blob/master/platformio.ini) (see #24) --> Including them directly is a bad idea, since it will increase the size of the repo by 3rd party libs.

I assume, we can close this issue.

boldt, 2020-09-09:
TODO: Add some notes to the github pages

amandel, 2020-11-15:
I guess this can be closed - there is already documentation for platform.io setup.

closed 2020-12-12 by amandel