openbikesensor/OpenBikeSensorFirmware #93: A lot of critical system functions have just void and no return value

Halbelunge, 2020-08-02:
Functions like void FileWriter::appendFile have just void.
They even check inside if something is wrong and then send it out via println().
There should be a return value (best as uint_8t, this just takes 1 byte on the stack) that is 0 if something has gone wrong and 1 if everything is ok.

Every function that calls a function with such a return value should return this value itself. If there is more then one call inside a function the return value should be the logical AND of all calls.

Example

[image]

amandel, 2020-12-12:
I've started to propagate the value to the caller. Still we need a way to show it to the user if such an error occurs - like data can not be written because of a write error or full buffer.

amandel, 2021-06-12:
Closing this, please create a new one if there is a concrete case. In general checking for return values was improved.

closed 2021-06-12 by amandel