openbikesensor/OpenBikeSensorFirmware #110: Ein paar Statistiken auf dem Display anzeigen

Natenom, 2020-09-09:
Wenn ich unterwegs bin, dann wäre es spannend, im Display folgendes zu sehen:
* Anzahl Überholvorgänge insgesamt (seit dem Einschalten)
* Anzahl Überholvorgänge > 2 m
* Anzahl Überholvorgänge > 1,5 (optional "UND < 2 m")
* Anzahl Überholvorgänge < 1 m
...

D. h. man könnte z. B. bei längeren Strecken auf Landstraßen das Gerät einmal ausschalten und einschalten und hätte am Ende der Landstraßenstrecke eine schöne Statistik dazu, wieviele Leute richtig überholt haben usw.

Ich fände das sehr interessant, weil man dann auch zwischendurch bei Touren/Fahrten darüber berichten könnte.

boldt, 2020-09-09:
Das Display hat so schon recht wenig Platz. Ich habe gerade keine Idee, wo man das noch unterbringen soll.

Natenom, 2020-09-09:
Ich benutze das Setup, dass ich nur den Abstandswert links sehe und Anzahl "drücken" + Anzahl "confirmed". Da, wo dann optional rechts der Wert nach rechts stehen könne, ist noch viel Platz.
Man könnte das als eigene neue Ansicht implementieren. Denn in der Regel ist nur der Abstand nach links wichtig in live.

boldt, 2020-09-09:
Ist definitv ein "enhancement", welches einiges an Arbeit benötigt.

ChrizKu, 2021-01-17:
denkbar wäre eine "Tabelle" mit 2 Zweilen:
oben, kleine Schriftart: `Σ | >2 | <1,5 | <1 | <0,7 `
darunter die Werte, größere Schrift.
Anzahl Werte je nach Platz…

evtl. im zeitlichen Wechsel mit gefahrene km und hh:mm.

Wann sind diese Infos interessant?
Während der Fahrt oder am Ende?
Beim Ausschalten oder bei Geschwindigkeit < x innerhalb der privacy zone?