openbikesensor/OpenBikeSensorFirmware #113: Verschieben in /upload funktioniert nicht, wenn Dateien schon vorhanden

Natenom, 2020-09-09:
Normalerweise läuft man nicht in diesen Fall rein, wenn man die Daten nur per WLAN hochlädt. Ich habe aktuell die Daten per SD-Karte ausgelesen und dabei auch die Datei tracknumber.txt gelöscht.

D. h. die erste Datei fängt wieder bei sensorData1.txt an.

So eine gibt es aber schon in /upload.

Lädt man jetzt die Daten hoch, dann passiert magic und man kann beliebig oft dieselben Daten hochladen.

boldt, 2020-09-09:
@uwewoessner

amandel, 2020-10-31:
This leads to multiple uploads of the same track - because the file is not deleted from the OBS and uploaded again in the next Upload.
Idea would be to try different naming in the target Folder in that case (add a .1, .2 ...)

uwewoessner, 2020-10-31:
duplicate names should not occur in the current version. If it still happens this should be fixed.

commit 1d341ee0 (2020-12-12): Change filename once we know the time (#173)

* Change filename once we know the time (fixes #95)
* Use a different name if the file is in upload already (fixes #113)
* Do not write file already with header creation to avoid empty files (fixes #62)
closed 2020-12-12 by amandel