openbikesensor/OpenBikeSensorFirmware #118: Create a new clean JSON for the configuration

boldt, 2020-09-16:


opatut, 2020-12-10:
Documentation could happen through [JSON Schema](https://json-schema.org/).

amandel, 2020-12-12:
I'd propose the following internal format, which will allow different presets based on a default preset: (yes I know there are no comments in Json)

For now only the "default" preset will be used but the structure is prepared to support presets. Not all values will be allowed to be overwritten in the presets, but to keep it simple I propose we treat all the same.

```json5
// Getting a value priority (privacyAreas will get merged):
// - value in "active" preset
// - value in "default" preset the one at ["obs"][0]
// - value hardcoded in firmware

{
  "obs": [
    { // 1st array entry is also the default
      "obsName": "MyOBS", // System-Default "OpenBikeSensor-<ID>"
      "numSensors": 2, // This is fixed, remove?
      "GPSConfig": 4, // Bitfield, saves memory could code it in stas for fix
      "satsForFix": 4, // 0 == no wait // -2 == wait for fix // -1 == wait for time // >0 == wait for number of sats
      "wifiSsid": "home-cloud",
      "wifiPassword": "verySecretThing",
      "name": "Wade",
      "offset": [ 30, 30],
      "portalUrl": "https://openbikesensor.hlrs.de",
      "portalToken": "3221432434234234001",
      "displayConfig": 524, // needs adopted display config!
      "confirmationTimeSeconds": 3,
      "privacyConfig": 2,  // see below privacyRecordPolicy
      "privacyRecordPolicy": "ALL", // |ALL|NO_CORDS|OVERTAKES_WITH_CORDS|OVERTAKES|NO_RECORDS|
      "bluetooth": true,
      "simRaMode": false,
      "devConfig": 0,
      "selectedPreset": "Art3mis", // store in separate file?
      "privacyArea": [
        {
          "lat": 48.7, "long": 9.1, "latT": 48.7, "longT": 9.1, "radius": 350
        },
        {
          "lat": 48.8, "long": 9.0, "latT": 48.8, "longT": 9.0, "radius": 351
        }
      ]
    },
    {
      "name": "Parzival",
      "offset": [ 30, 30],
      "portalToken": "3221432434234234001"
    },
    {
      "name": "Art3mis",
      "offset": [ 32, 32],
      "portalToken": "3221432434234234002",
      "confirmationTimeSeconds": 4,
      "privacyRecordPolicy": "NO_RECORDS",
      "bluetooth": false
    }
  ]
}
```

commit c74769f0 (2020-12-24): New config structure, load & save

Fixes #118
commit 3ec934f3 (2020-12-24): New config structure, load & save (#177)

- Create and document a new clean JSON for the configuration fixes #118
- Config format is structured json this allows:
  - PrivacyAreas are stored as array inside Json (not a properties with counter in the property name)
  - SensorOffstes are stored as array inside Json
- Main config is inside a array which is prepared to hold different config profiles for certain values.
- Configuration file is not re-written at each OBS restart, only ServerMode writes config
- A backup of the configuration is created before writing the new config (might be dropped a [... cut]
closed 2020-12-24 by amandel