openbikesensor/OpenBikeSensorFirmware #122: Firmware auf ESP32 Board übertragen klappt nicht

Leom-hue, 2020-09-24:
Hallo, 

leider habe ich keine Erfahrung mit Programmieren so dass ich mir selbst nicht helfen kann (Google konnte mir auch nicht helfen). Die Schritte um die Firmware auf das Board zu übertragen (mit VSCode), habe ich so ausgeführt wie beschrieben, leider erscheint beim Upload die Fehlermeldung 

Changing baud rate to 460800
Changed.
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Compressed 15872 bytes to 10319...
A fatal error occurred: Timed out waiting for packet content

Hat hier jemand schon mal ähnliche Erfahrungen gemacht oder kann mir noch weiterführende Tipps/Einstellungen geben, womit es klappen könnte?

Gruß Leo

Leom-hue, 2020-09-25:
GELÖST: Der ESP32 war bei mir zunächst bereits verlötet. Das Übertragen der Firmware hat dann nicht geklappt. Nach erneutem Entfernen aller Verkabelungen, hat das Übertragen einwandfrei funktioniert.

boldt, 2020-09-27:
@Halbelunge Hast Du eine Idee, was hier passiert sein kann? Das Gerät sollte ja eigentlich immer zu flaschen sein.

amandel, 2021-01-03:
I suggest this can be closed.

closed 2021-01-13 by amandel