openbikesensor/OpenBikeSensorFirmware #123: getDistance vs. getDistanceSimple

amandel, 2020-10-09:
I try to dig a bit into the code and stumbled upon https://github.com/Friends-of-OpenBikeSensor/OpenBikeSensorFirmware/blob/bc806f2e5978dc203d8c2d14d70a1df45d24c7dc/OpenBikeSensorFirmware/sensor.cpp#L93 vs. https://github.com/Friends-of-OpenBikeSensor/OpenBikeSensorFirmware/blob/bc806f2e5978dc203d8c2d14d70a1df45d24c7dc/OpenBikeSensorFirmware/sensor.cpp#L114

It feels that 'getDistance' is overly complicated and makes further changes hard. Can this be cleaned up or am I missing something (besides some code dealing with floats, most code can e replaced by using 'pulseIn')?

boldt, 2020-10-10:
Yes, it's very complicated.

@tobst - we discussed this some time ago. So you remember the conclusion?

I tried to simplify this some time ago as well, bit it didn't work.

See also #73

tobst, 2020-10-10:
The code I wrote was wirh the intention of parallel triggering of the sensors to increase the measureing rate. It revealed some problems on the street. Further research needed if possible. If only serial measuring is possible, the complicated code is obsolete and pulseIn() should be sufficient.

amandel, 2020-10-10:
Thanks for the feedback. I found a implementation in the net for parallel measure that uses interrupts. If this is OK, I can give it a try to implement this for our case.  This would also avoid that the pulse of one sensor reaches the other one (however this could happen). But since we anyway just loop and wait for results I'm not sure if it is needed. Nevertheless it sounds interesting to implement.

I would also like to implement that only the median of the last 3 (5?) measures is taken, what do you think? Is there a error rate seen in the wild?

I found some math in the net that we have a "high" air temperature dependency. When I'm correct this is in the area of +/- 5cm is this fine (sorry, might already have been discussed) see also:

https://github.com/amandel/OpenBikeSensorFirmware/blob/319e7059ef7046808432e8bbe140e298eb960de4/OpenBikeSensorFirmware/sensor.cpp#L23-L45

boldt, 2020-10-10:
@Halbelunge - we discussed that temperature dependency already...

amandel, 2020-10-10:
I've now have also created a interrupt based version that allows both sensors to be read in parallel. With the delay between reads of 75ms we can read the distance up to 13 times per second. Is this sufficient? I still wait for result or timeout, this might be further improved - not to increase the measurement frequency but to have more time for other tasks.

I hope thee coding & style is OK. I'm more into Java so I hope not to hit to many c++ pitfalls.

amandel, 2020-10-11:
I've created a PR: #124

commit 442e3efc (2020-10-13): Simplify getDistance - fixes #123
closed 2020-10-13 by amandel