openbikesensor/OpenBikeSensorFirmware #168: Upload of long Tours

mega-stoffel, 2020-12-04:
just today I rode a 120km tour, 6,5h. The sensorData.csv is 11,5MB.
I uploaded the csv to the HLRS portal and would have expected, that the file was moved to the uploaded directory.
But this didn't happen. There is no file with this name, therefore I assume, it's due to its size.

amandel, 2020-12-08:
Seems the file is not closed after the transfer - this is more likely the reason. I'll check with the other file handling changes #95 etc.

opatut, 2020-12-12:
We're storing the file in the mongodb for now, as a string inside a JSON document -- which is somewhat dumb ;)

It will be solved by openbikesensor/obsAPI#18.

boldt, 2021-03-29:
@opatut what's the staus on this?

opatut, 2021-03-30:
I'm considering to rewrite, again, the whole data logic. I am trying to invoke the python (FACE) scripts from the API instead of parsing the files in JS, since the python code is way more advanced.

I think also that I'll make a clean cut, manually migrate the existing database, and then have a totally different storage format for the infromation we need. It is a mess right now, and I kind of don't want to deal with writing an automatic migration from that... if only I had more time ;)