openbikesensor/OpenBikeSensorFirmware #169: Need bulk configuration mode that allows usable and fail safe configuration of many (~20 / ~30) OBS units

amandel, 2020-12-06:
@tobst can you describe the needed parameters / intended use case?

tobst, 2020-12-06:
Intended use case: A test coordinator has many sensors and just as many persons to equip with those devices. They all have different offset widths, some have WiFi, some have accounts on openbikesensor.hlrs.de
The coordinator has a list of device identifiers (mac-adresses) and config values to apply to those devices. 
If a full configuration is applicable, it would be nice, if device holders had a chance to obtain a configuration file without a device an supply it to the test coordinator.

amandel, 2021-01-21:
If you have the SD card at hand #217 might be worth a look

amandel, 2021-07-05:
@opatut   - this is implemented somewhere, right?

opatut, 2021-07-06:
We have some options for this. In general, [obs-privision](https://github.com/openbikesensor/OpenBikeSensor-Scripts/blob/main/obs/bin/obs_provision.py) is intended to be a wrapper script for all kinds of tools that relate to managing multiple devices conventiently. Note that the script currently only supports one operation, bulk download. More to come.

* If the devices can join the same network, the script could send the new configs OTA.
* If the devices are one-by-one connected via USB, [flashing the config directly](https://gist.github.com/opatut/c1690ad5b79f6c46fe441b192fb38bf3) is an option, too.
* At some point we might want to add PC-OBS communication via serial USB to allow some features that are otherwise hard to do, such as:
  - triggering test routines and reading the result directly
  - sending config files while the device is running, without flashing
  - reading the device ID
* We might write some helper scripts to map maybe from a CSV or XSLX file with some sub fields to a bunch of JSON config files, stored in folders named after the device ID. That could be used by obs-provision to look up the correct config for devices that should *not* share the same configuration. Maybe some existing templating engine could be used instead, too, to save some effort.
* If you want to go crazy, building a graphical frontend for obs-provision would also be possible.