openbikesensor/OpenBikeSensorFirmware #181: Automatically upload track data once connected

amandel, 2020-12-23:
Offer as an opt out(!) configuration that the OBS automatically uploads all available track data once it "feels" a internet connection. 

It should be on by default (might be when the portal token is entered?). 

Use case is to ease up the typical steps needed to upload the data - without the need of a 2nd device. Once configured it allows to do the upload just by starting OBS in Wifi-Mode. So it also would be nice to see some progress information in the display during the upload.

Originally from @tobst

tobst, 2020-12-23:
Default on/off depends on personal preferences.
The chance to cancel uploading or not even start might be usefull in case you just want to edit the config real quick before riding (switched bike etc) and are still in wifi range.

amandel, 2021-01-02:
I've now selected a 5 second button press that can directly start the upload after start-up of the OBS in server mode.
This needs a connected display, that should be the normal case for today. Also the display is used for upload progress detail information.

closed 2021-01-03 by amandel