openbikesensor/OpenBikeSensorFirmware #190: How to implement an encryption for the wifi password?

Halbelunge, 2020-12-30:
How can we implement a encryption for the wifi password without storing a private key and without entering a password each time.

Store the private key in EEPROM? and only the encrypted PW on the SD card?

I look for a RSA encryption lib.

amandel, 2020-12-30:
See also #119

mega-stoffel, 2021-01-01:
and furthermore: would it be possible to use a different type of "input-field"?
From early web-form programming I learned, there's a "password" input type. Entries in these forms are not stored in the browser's AutoFill. But by using the OBS config my WIFI password is stored in the mobile Chrome browser's AutoFill.

amandel, 2021-01-02:
@mega-stoffel  - the input is of type password and so should be treated as your browser treats every password (typically by not displaying the input and treating it as secret). Can you point to the concrete field you refer to and the browser you use?

amandel, 2022-05-08:
Actually the password is not stored on the SD card (it never was) but in the ESP internal flash memory. It is exposed in clear-text with the configuration backup.