openbikesensor/OpenBikeSensorFirmware #193: link after upload

mega-stoffel, 2020-12-31:
in the confirmation screen, which shows all uploaded tracks, there could be a link to the upload-site.
Directly after I upload tracks, I usually got there (right now manually) and look at the tracks and make modifications.

amandel, 2021-01-02:
Question to the API side. Can we return a direct link to the track on the portal as response to the upload? Otherwise it would be a static link to the configured portal.

Steffeng5, 2021-01-02:
As you Upload multiple Tracks at once: We can maybe return a link for each uploaded track, do you then want to list all Tracks with links?

amandel, 2021-01-02:
There is one multi-part HTTP POST request per track from OBS to the API, the idea is that with the 200 status code you also send a URL as text body or might be as HTTP location response header (change status code to 201?). Not sure how to represent the links then on the UI. Might be with the OK status symbol but we will see.
The user can then click through all uploaded track one by one. (links should open in new window)

amandel, 2021-01-08:
@Steffeng5  -  die v0.4.391 ist auf einen location http header mit link auf das portal vorbereitet.