openbikesensor/OpenBikeSensorFirmware #205: Report of Time and number of sattelites does collide

Halbelunge, 2021-01-09:
This was introduced somewhen in the main.

} else {
      char timeStr[32];
      snprintf(timeStr, sizeof(timeStr), "%02d:%02d:%02d %dsa",
               gps.time.hour(), gps.time.minute(), gps.time.second(), gps.satellites.value());
      satellitesString = String(timeStr);
    }
    displayTest->showTextOnGrid(2, 5, satellitesString,DEFAULT_FONT);
    
    The problem is, that if a time is there, but 0 sattelites, the time is displayed but the number of satellites are out of bounds.
    there Should be a check, or everyhing should scroll 1 row up.

amandel, 2021-01-09:
Not sure where you see the collision? Display of time is by intention - the time is displayed while waiting for satellite reception.

Halbelunge, 2021-01-09:
cant get GPS signal at the moment, so I cant provide a picture of the dispaly contents

bit it trys to display xx:xx:xx 0 sats but can only display xx:xx:xx 0 sa 
and the a of sa is only displayed half.

I have (for now) implemented a small auto scrolling that scrolls all lines one up, the moment a time is found.
So the following is shown.

xx:xx:xx
0 / 4 sats

instead of 
xx:xx:xx 0 sats

Halbelunge, 2021-01-09:
This collides ith your pull request Andreas :/

I try to get a picture of what is the problem

Halbelunge, 2021-01-09:
[image]
This is what I talk about

amandel, 2021-01-09:
Actually we could overwrite or skip the csv line. The actual file is not written at that point, just some buffers are prepared in memory. So this can hardly fail and it gives no real information. 
Then we can print something like
'0 Satellites' in the 2nd line below the time.

amandel, 2021-01-14:
@Halbelunge should be OK with the new release

closed 2021-01-21 by amandel