openbikesensor/OpenBikeSensorFirmware #216: Privacy Area should be defined as irregelar polygon instead of circle

stberg-os, 2021-01-20:
If the privacy area is defined by center and radius it still allows guessing the center point of the privacy area by viewing all tracks of a user.

If the privacy area is an irregular polygon as shown in the image, it's impossible to guess the actual point.
[image]

amandel, 2021-01-22:
The filtering of the privacy area does not happen on the center of the circle. Instead as @mega-stoffel  explained the circle is shifted by a random offset. If there is no strong need I would *not* add additional complexity to this, neither on the software nor on the user / usage side. There is a more sophisticated filter possibility planned on the portal side, see https://www.openbikesensor.org/user-guide/privacy-zones.html

amandel, 2021-07-23:
Closing for now as described above.

closed 2021-07-23 by amandel