openbikesensor/OpenBikeSensorFirmware #219: Reset display every ~60sek

Halbelunge, 2021-01-25:
To give better user experience reset the display every ~60 seconds to give bring a glitched display back to normal

Halbelunge, 2021-01-25:
just do an m_display->init() is not a good way, because with this the buffers will be reset in the allocation process. We need to skip this.

all that is left is an
m_display->sendInitCommands();
m_display->resetDisplay();

But sadly these 2 functions are in the "protected" part of the OLEDDisplay.h if we could take out this "protected" at that point, we can just run this as a light reset without allocating the buffers again.

amandel, 2021-07-05:
Idea could be to react to i2c errors with this display refresh and leave the display alone otherwise