openbikesensor/OpenBikeSensorFirmware #225: Config server must use https

amandel, 2021-02-18:


opatut, 2021-02-19:
> check if we can get a setup running leveraging lets encrypt and our domain.

Impossible. Well, technically this might work, but it defies the whole purpose. Sharing one certificate is not possible because that would be public and basically be worthless, plus quickly rejected by letsencrypt and also agains their terms of use. So such a feature would inevitably involve a certificate per OBS. But still, this is private key in the hands of the OBS user, who can always extract it from the OBS and start impersonating other devices. You cannot secure it against extraction from the device. And any infrastructure you build to auto-generate per-device certificates will be a security nightmare and is almost definitely against letsencrypt's terms of use. There is now way you can make a secure server on a device you don't have control over, and make "the public", aka everybody who connects to it, believe it is trusted .That defies the whole point of certificate validation, if you hand them out like candy.

I think our best bet is to use a self-signed certificate on the device, one per device. Then the device could display its fingerprint on the screen for you to verify when adding an exception in your browser.

amandel, 2021-02-19:
Actually the 2nd article describes how it was done. As you wrote to effort is high and a 'security-nightmare'.
Let's do one thing after the other. I'm already not sure if we have enough space for the generation of the key on the device. Might be, even the https lib needs to much code space. We will see.

opatut, 2021-02-19:
> Actually the 2nd article describes how it was done.

Yes, but that only works because the servers on the internal networks AND the internal DNS were "under control" of the same entity that also generated the certificates from publicly assigned DNS entries on publicly reachable servers. There, no attacker (supposedly) could steal the certificates/keys from the internal servers. Which we cannot provide, the devices are cycling around in cities. Furthermore, it requires a central internet-accessible server that the device talks to, which we also do not have during initial setup, where you enter your wifi password, which is already a bad start. Lastly, it would mean that each device would need its own subdomain, and then it'd have to prove to the server on the next time it wants to update its certificate that it is still the same device, otherwise I could request an update for your device's subdomain. The server could however not verify the device's authenticity, just as little as the users browser can. In the end it is a bunch of catch-22's stacked together, where nobody has initial trust ;)

> Let's do one thing after the other.

I agree :blush: Best-effort TLS with a self-signed certificate will already be a big improvement. Also we could allow users to bake custom certificates into their firmware if they want, through developer options. That'll be great for people who want to locally provision some devices manually without users getting security warnings, such as in some of the research projects going on. They could very well do this DNS-rebind trick, if they set up a DNS in the network where they configure the devices. Probably not many people will do this, but it'd be nice to have, e.g. for developers too.

amandel, 2021-02-19:
Next topic is diet. Just pulling in the fhessel/esp32_https_server lib:
````
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  22.3% (used 72924 bytes from 327680 bytes)
Flash: [==========]  104.7% (used 2058365 bytes from 1966080 bytes)
Error: The program size (2058365 bytes) is greater than maximum allowed (1966080 bytes)
*** [checkprogsize] Explicit exit, status 1
````
I'm not sure if we are better off using the espressif libs.

opatut, 2021-02-19:
Does this analyzer give useful information? Do you have an idea where our memory is spent? If it's the stupid HTML literals, we could consider starting to offload them ...

amandel, 2021-02-19:
I did not look into this yet. I'm positive to find potential candidates - we carry some fonts in the memory etc. but 1st I want to merge in the GPS changes. and may be the parallel TOF reading. This will also need some more memory. 
Bluetooth eats most of the memory, might be we can drop of some parts of this lib - we are only using BLE Server part.

amandel, 2021-03-20:
We use a static self signed cert. The same cert is used for all obs. While it allows attackers to fake an obs server it still allows us to encrypt the traffic. We might mitigate this by displaying a secret on the obs display as well as the obs html pages. (TODO)

closed 2021-03-22 by amandel