openbikesensor/OpenBikeSensorFirmware #237: Komische Werte beim Invertieren des Displays

boldt, 2021-04-02:
Bug aus dem Slack: Firmware = v 0.4.496

> Dann bin ich auf die Idee gekommen das Display umzustellen auf schwarze Schrift vor weißem Hintergrund. Ab diesem Zeitpunkt zeigt das Display für den Abstand links beliebige Werte die zumeist zwischen 1 Meter und 2,5 Meter liegen, auch wenn ich den Sensor senkrecht in den freien Himmel ausrichte. Nach Rückstellen des Displays auf weiße Schrift vor schwarzem Grund habe ich wieder Normalfunktion. Das lässt sich beliebig oft wiederholen.

amandel, 2021-05-30:
@Halbelunge hast du eine Idee dazu? An der Software dürfte das nicht liegen, wenn ich das auch bisher nicht mit der 0.4.496 geprüft habe. Könnte die i2c Bus Kommunikation für solche Störungen sorgen?

amandel, 2021-09-18:
I slack we also got one report of the opposite behavior - the distance measurement was more stable with inverted display!?