openbikesensor/OpenBikeSensorFirmware #240: Wrong Information about Linux Devices in "Initial ESP flashing" Manual

EarthlingX, 2021-04-26:
At least on Debian systems (checked here with Debian Buster), the recommended workflow to flash the ESP32 is **not** correctly described [in the manual](https://github.com/openbikesensor/OpenBikeSensorFirmware/blob/master/docs/software/firmware/initial_flash.md).

**I had to get it work this way:**

1. Install package _esptool_ via `apt-get update` and afterwards `apt-get install esptool`
2. Navigate into directory where you have extracted the zip (e.g. _/home/user/Downloads/OBS_)
2. Copy and paste exactly this command (and nothing else) into the shell and run with "enter": `esptool \
    --chip esp32 \
    --port /dev/ttyUSB0 \
    --baud 921600 \
    --before default_reset \
    --after hard_reset \
    write_flash -z \
    --flash_mode dio \
    --flash_freq 40m \
    --flash_size detect \
    0x1000 0x01000.bin \
    0x8000 0x08000.bin \
    0xe000 0x0e000.bin \
    0x10000 0x10000.bin`

The following flashing process worked pretty well and succeeded in all cases. However, the initial approach mentioned in the manual (with its first line `python3 esptool.py \`) did never work, it always led to "Command/file not found" and similar output in the shell prompt. Reason for this could be that the file "esptool.py" was missing in the zip file?

derSammy, 2021-04-26:
Thanks for this feedback. I face the same issue with the non-successfull Linux instruction (using Linux Mint) and solved it equivalently by (having Python3 (incl. Pip) installed) by
`sudo pip3 install esptool`
and than
`esptool.py \
  --chip esp32 \
  --port /dev/ttyUSB0 \
  ...`
Anyway, it seems to be necessary to install esptool with root privileges.

SubOptimal, 2021-04-27:
My suggestion would be not to provide a distro-specific command as general advice.

- leave the command in the documentation as is; it would support users not familiar with Linux to know for which tool to search for their distribution (`esptool.py`) and (if needed) to install Python 3 (and not Python 2)
- to provide even more help, we could add a hint section for some Linux distributions to reflect the differences 

**Debian**
- as root `apt-get install esptool`
- call `esptool` instead of `python3 esptool.py`
-`/usr/bin/esptool` is a soft link to `/usr/share/esptool/esptool.py`, which the Python 3 interpreter executes

**Arch**
- as root `pacman -S esptool`
- call `esptool.py` instead of `python3 esptool.py`
- `/usr/bin/esptool.py` is the script itself, which the Python 3 interpreter executes

We might provide a general solution for distributions not providing a package to install `esptool.py`. It seems Mint is such a case. On the website http://packages.linuxmint.com/ no package is found for the term "esptool".
Here we should avoid suggesting the solution `sudo pip3 install esptool` as this can break the package manager of the distribution.

opatut, 2021-04-27:
We could also leave a hint that, if you don't mind the bigger disk size it requires, compiling yourself and flashing through `platformio` is rather convenient and pretty much failsafe. 

* install platformio tools for your distro
* clone the firmware repo
* run `pio run -t upload` from the repository

amandel, 2021-12-11:
With https://install.openbikesensor.org/ no platform specific setup for the initial installation is needed any more. So I close this.

closed 2021-12-11 by amandel

boldt, 2021-12-12:
Should we add https://install.openbikesensor.org/ to https://www.openbikesensor.org/docs/hardware/v00.03.12/build-instructions/#schritt-18-elektronik-testen as well?

amandel, 2021-12-12:
Is the install / flash part not covered at all? Sure, why not.

opatut, 2021-12-13:
Now it is: https://openbikesensor.org/flash

We should just add a fourth section there, IMO, `Flash from Browser`