openbikesensor/OpenBikeSensorFirmware #241: Implement new OAuth flow

opatut, 2021-04-30:
Starting v0.2, the Portal will have a proper OAuth Authorization Code flow implemented and we can use that in the configserver to connect a user account. It allows us to have a simple "Log in with OpenBikeSensor Portal" button instead of having to copy-paste API Keys.

This is a bit of code, and can be implemented in 95% JavaScript for developer convenience. It will require:

* A JS file to include in the login flow pages
* A redirect endpoint that handles the login
* A configuration of the client id and auth server URL, for which we should compile in the defaults
* A refresh token will be issued that has to be exchanged for an access token before uploading to the portal, so one additional request before the uploading can begin. The access token is valid for an hour or so, so it can be used for all subsequent uploads.

This does not yet work on the currently deployed HLRS instance, but that one is deprecated anyway. We can already start writing this component and release it when we have the new portal up and running.

Steffeng5, 2021-05-14:
Would it still be possible to add a simple api token if this is implemented? We have to get sure that this can also be provisioned on many devices without user interaction via predefined config file on SD card

opatut, 2021-05-14:
It doesn't matter where the refresh-token comes from. It will end up in config, so you can feed it through the login "wizard" in the browser or sideload it through config on SD.

But you shouldn't. Each device should have its own token, so you can invalidate them independently. If you're logged in to the portal on the device you use for configuration, then logging in the device is a one or two click process. Probably easier even than finding the refresh token and moving it to some config file in the right format.

We also have `obs-provision` script in the scripts repo which will at some point be able to deal with mass configuration, even be considered its own client for issuing tokens and then every device gets its own token without any clicking.