openbikesensor/OpenBikeSensorFirmware #262: Regression: Noisy measurement on right sensor with beta

gluap, 2021-07-12:
When using the stable firmware version and I measure into the open sky I reliably measure a distance of "999" indicating that the sensor is not receiving an echo (as is expected) [video](https://user-images.githubusercontent.com/44007906/125313736-d3232600-e335-11eb-961a-b108349751a9.mp4). In the beginning of the video there are a few non-infinity measurements that reflect me positioning the case such that the right sensor points skywards.

When using the 0.8.630 beta version, I reliably see intermittent distances measured appear on the display. [video](https://user-images.githubusercontent.com/44007906/125313729-d1596280-e335-11eb-8e2d-a247cfbbec88.mp4)

In both cases bluetooth is disabled.

Both videos were taken right after one another: first video 0.8.630, then firmware reflashed with 0.7.596 and right away video with 0.7.596. The position of the sensor is also the same.

.

amandel, 2021-07-12:
Thanks for digging this down. Does it still get false measures when you cover the sensor with  a packet of paper handkerchiefs or some fabric?

gluap, 2021-07-12:
- Yes, same behaviour with a piece of cloth covering the sensor (one layer - several layers, no difference).
- Only the right sensor is affected.
- Noise seems to only be measured when measuring infinity. Indoors measuring between two walls less than the max distance away, both values are stable.
- Additionally sometimes it seems like the right sensors measured value freezes from time to time (even when measuring actual distances). Subjectively the value freezing only happens in connection with noise measurements, but hard to say as during stable measurements the display is also frozen. This behaviour is also limited to the right sensor
- None of the above occurs with the 0.7.596 firmware.

gluap, 2021-07-12:
@amandel reply to edit: in both videos I'm sitting in a garden chair next to the table holding the obs in the other hand. I may have shifted seating position between the two videos so 70cm (processed value) is likely sensor to ground while 10 cm is sensor to table.  The issue is reproducible independent of where I use the obs, as long as the left sensor would measure infinity it shows this behavior.

amandel, 2021-07-12:
Thanks for the details and the quick answer. I saw your response after my edit :/.

gluap, 2021-07-12:
@amandel In case there are remaining doubts I have two longer videos demonstrating the effect with both firmware versions.

PR #263 (2021-07-22): Refine tof sensor timing
- be more precise for the high signal timing by sending the trigger per sensor
- count all special cases by increasing numberOfNoSignals, just to get a feeling
- add some debug output to the csv file
- should close #262
closed 2021-07-22 by amandel