openbikesensor/OpenBikeSensorFirmware #272: Display recorded value in inverted display

wondalfo, 2021-09-11:
If I look on the display, it is in most cases, shortly after pressing the button. The I see arbitary values in the inverted display. Can you change it like this, that the recorded value is displayed.

amandel, 2021-09-18:
I see your point and we had this behavior before. I changed it when there was demand to be able to directly confirm the next overtaking after the button press. I'm thinking about changing how the display is implemented, which would also allow to display different things like the current measurement together with the confirmed measurement instead of what is currently the "inverted" display. 
Would this cover your idea?

wondalfo, 2021-09-25:
> I see your point and we had this behavior before. I changed it when there was demand to be able to directly confirm the next overtaking after the button press. I'm thinking about changing how the [...]

Probably yes.