openbikesensor/OpenBikeSensorFirmware #275: Enable initial installation leveraging ESP web tools

amandel, 2021-11-05:
With the [ESP web tools](https://esphome.github.io/esp-web-tools/) we can allow initial installation (aka flashing) via web browser. This can even be extended to include WiFi configuration once [improv](https://www.improv-wifi.com/) is released working via serial.

Problem is that the installation needs a rather static web page with some options, a manifest file, and the needed binaries. The binaries can be external (url reachable) resources but the urls must be part of the `manifest.yaml`.

```yaml
{
  "name": "OpenBikeSensor",
  "builds": [
    {
      "chipFamily": "ESP32",
      "improv": false,
      "parts": [
        { "path": "0x01000.bin", "offset": 4096 },
        { "path": "0x08000.bin", "offset": 32768 },
        { "path": "0x0e000.bin", "offset": 57344 },
        { "path": "0x10000.bin", "offset": 65536 },
        { "path": "https://github.com/openbikesensor/OpenBikeSensorFlash/releases/download/v0.1.22/flash.bin", "offset": 3735552 }
      ]
    }
  ]
}
```

I tried to put all in the release page, but it seems we can not add script content into the GitHub rendered page. So we might add the page to the documentation area. Adding all `*.bin` and the `manifest.yaml` as release of the OpenBikeSensorFirmware artifact would allow us to create a static page that uses JavaScript to select the intended download version (latest). I need to mention that we also need to set some cors HTTP headers for this setup to work.

opatut, 2021-12-02:
See also openbikesensor/portal#125.

amandel, 2021-12-08:
See https://github.com/openbikesensor/OpenBikeSensorFlasher for the imlementation

closed 2021-12-08 by amandel