openbikesensor/OpenBikeSensorFirmware #287: please create upload logfile on SD-Card

Ventusfahrer, 2021-12-31:
currently there is no logfile created on the SD-Card which contains the upload status of the last (or all) upload attempts.
At least I did not find any in the part which is visible in the _Show SD Card Contents_ directories of OBS-Unit.
At the end of an upload process initiated by pressing the hardware button you just get the information which is shown on the bottom of the screen
[image]

There are various reasons why the upload may fail:
- no or wrong Upload Url
- no wrong API key
- no connection to the upload host
- breaking connection during upload
- file was already uploaded to the OBS-server

In order to ease finding the root cause for failed uploads it would be very helpfull to have a logfile with detailed error messages.

amandel, 2021-12-31:
If you do the upload using the web ui, you can see the message, if you hover over the error indicating red cross, as a tool-tip.  

Related to #96

gluap, 2022-01-02:
Also: If this feature is implemented it makes sense to also have the time of the failed uploads in the logs - Allowing to potentially cross-check with server side logs.

amandel, 2022-01-07:
Could you already detect the cause by using the web ui for the upload?