openbikesensor/OpenBikeSensorFirmware #302: Links for development setup are dead

FabianSchwartau, 2022-04-19:
Hi,
the links in the readme to the setup guides for vscode/clion/arduino are dead.
If you want people to help improving stuff, this information would be nice to have ;)
BTW: Where did that information move? A quick search did not reveal it. Now I am stuck and also annoyed after just 60 seconds into "development" :(

boldt, 2022-04-19:
You're right, broken links are always annoying.
I'll try to figure out, where the docs moved to.

boldt, 2022-04-19:
It seems, that the develoment setup never moved to the new web site. Here they are:

https://github.com/openbikesensor/archive.openbikesensor.github.io/blob/old-jekyll-website/software/firmware/setup.md

@amandel Shall we move the setup into this repo or back to the main docs?

amandel, 2022-04-20:
Actually I do not know why it was not moved over. If it is to technical it might be placed in https://github.com/openbikesensor/OpenBikeSensorFirmware/tree/master/docs/software/firmware ?
@FabianSchwartau  - be warned I'm not sure how current this documentation is. Participation is very welcome!

FabianSchwartau, 2022-04-20:
No worries, I figured it out myself, but took me longer than it should have ;)