openbikesensor/OpenBikeSensorFirmware #305: Build fails due to missing `hwcrypto/sha.h`

tfar, 2022-05-06:
I am trying to build and flash the firmware. Following the instructions on https://github.com/openbikesensor/archive.openbikesensor.github.io/blob/old-jekyll-website/software/firmware/setup.md . Pressing the `Default -> General -> Build` task in the PlatformIO sidebar leads to the following errors:

```
> Executing task in folder OpenBikeSensorFirmware: platformio run <

Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (4.2.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.20002.220503 (2.0.2) 
 - tool-esptoolpy 1.30300.0 (3.3.0) 
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 39 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Adafruit BusIO> 1.11.5
|   |-- <Wire> 2.0.0
|   |-- <SPI> 2.0.0
|-- <ArduinoJson> 6.19.4
|-- <CircularBuffer> 1.3.3
|-- <ESP8266 and ESP32 OLED driver for SSD1306 displays> 4.3.0
|   |-- <Wire> 2.0.0
|   |-- <SPI> 2.0.0
|-- <Adafruit BMP280 Library> 2.6.2
|   |-- <Adafruit Unified Sensor> 1.1.5
|   |-- <Adafruit BusIO> 1.11.5
|   |   |-- <Wire> 2.0.0
|   |   |-- <SPI> 2.0.0
|   |-- <Wire> 2.0.0
|   |-- <SPI> 2.0.0
|-- <VL53L0X> 1.3.1
|   |-- <Wire> 2.0.0
|-- <esp32_https_server> 1.0.0+sha.253013d
|-- <HTTPClient> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <WiFiClientSecure> 2.0.0
|   |   |-- <WiFi> 2.0.0
|-- <SD> 2.0.0
|   |-- <FS> 2.0.0
|   |-- <SPI> 2.0.0
|-- <Update> 2.0.0
|-- <WiFiClientSecure> 2.0.0
|   |-- <WiFi> 2.0.0
|-- <SPIFFS> 2.0.0
|   |-- <FS> 2.0.0
|-- <ESP32 BLE Arduino> 2.0.0
|-- <FS> 2.0.0
|-- <DNSServer> 2.0.0
|   |-- <WiFi> 2.0.0
|-- <ESPmDNS> 2.0.0
|   |-- <WiFi> 2.0.0
|-- <WiFi> 2.0.0
|-- <HTTPUpdate> 2.0.0
|   |-- <HTTPClient> 2.0.0
|   |   |-- <WiFi> 2.0.0
|   |   |-- <WiFiClientSecure> 2.0.0
|   |   |   |-- <WiFi> 2.0.0
|   |-- <Update> 2.0.0
|   |-- <WiFi> 2.0.0
Building in release mode
Compiling .pio/build/esp32dev/src/Firmware.cpp.o
Compiling .pio/build/esp32dev/src/OpenBikeSensorFirmware.cpp.o
Compiling .pio/build/esp32dev/src/VoltageMeter.cpp.o
Compiling .pio/build/esp32dev/src/bluetooth/BatteryService.cpp.o
In file included from .pio/libdeps/esp32dev/esp32_https_server/src/HTTPSServer.hpp:11,
                 from src/configServer.h:37,
                 from src/OpenBikeSensorFirmware.h:32,
                 from src/OpenBikeSensorFirmware.cpp:26:
/Users/tfar/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/openssl/include/openssl/ssl.h:8:2: warning: #warning "OpenSSL component will be removed from ESP-IDF in v5.0, please use esp_tls instead" [-Wcpp]
 #warning "OpenSSL component will be removed from ESP-IDF in v5.0, please use esp_tls instead"
  ^~~~~~~
In file included from .pio/libdeps/esp32dev/esp32_https_server/src/ConnectionContext.hpp:8,
                 from .pio/libdeps/esp32dev/esp32_https_server/src/HTTPRequest.hpp:10,
                 from .pio/libdeps/esp32dev/esp32_https_server/src/HTTPSCallbackFunction.hpp:4,
                 from .pio/libdeps/esp32dev/esp32_https_server/src/ResourceNode.hpp:7,
                 from .pio/libdeps/esp32dev/esp32_https_server/src/HTTPServer.hpp:20,
                 from .pio/libdeps/esp32dev/esp32_https_server/src/HTTPSServer.hpp:15,
                 from src/configServer.h:37,
                 from src/OpenBikeSensorFirmware.h:32,
                 from src/OpenBikeSensorFirmware.cpp:26:
/Users/tfar/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/openssl/include/openssl/ssl.h:8:2: warning: #warning "OpenSSL component will be removed from ESP-IDF in v5.0, please use esp_tls instead" [-Wcpp]
 #warning "OpenSSL component will be removed from ESP-IDF in v5.0, please use esp_tls instead"
  ^~~~~~~
In file included from .pio/libdeps/esp32dev/esp32_https_server/src/HTTPResponse.hpp:12,
                 from .pio/libdeps/esp32dev/esp32_https_server/src/HTTPSCallbackFunction.hpp:5,
                 from .pio/libdeps/esp32dev/esp32_https_server/src/ResourceNode.hpp:7,
                 from .pio/libdeps/esp32dev/esp32_https_server/src/HTTPServer.hpp:20,
                 from .pio/libdeps/esp32dev/esp32_https_server/src/HTTPSServer.hpp:15,
                 from src/configServer.h:37,
                 from src/OpenBikeSensorFirmware.h:32,
                 from src/OpenBikeSensorFirmware.cpp:26:
/Users/tfar/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/openssl/include/openssl/ssl.h:8:2: warning: #warning "OpenSSL component will be removed from ESP-IDF in v5.0, please use esp_tls instead" [-Wcpp]
 #warning "OpenSSL component will be removed from ESP-IDF in v5.0, please use esp_tls instead"
  ^~~~~~~
In file included from .pio/libdeps/esp32dev/esp32_https_server/src/HTTPServer.hpp:23,
                 from .pio/libdeps/esp32dev/esp32_https_server/src/HTTPSServer.hpp:15,
                 from src/configServer.h:37,
                 from src/OpenBikeSensorFirmware.h:32,
                 from src/OpenBikeSensorFirmware.cpp:26:
.pio/libdeps/esp32dev/esp32_https_server/src/HTTPConnection.hpp:9:10: fatal error: hwcrypto/sha.h: No such file or directory
 #include <hwcrypto/sha.h>
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/esp32dev/src/OpenBikeSensorFirmware.cpp.o] Error 1
In file included from src/VoltageMeter.cpp:24:
src/VoltageMeter.h:34:32: error: 'ADC1_GPIO34_CHANNEL' was not declared in this scope
       adc1_channel_t channel = ADC1_GPIO34_CHANNEL);
                                ^~~~~~~~~~~~~~~~~~~
src/VoltageMeter.h:34:32: note: suggested alternative: 'ADC1_CHANNEL_7'
       adc1_channel_t channel = ADC1_GPIO34_CHANNEL);
                                ^~~~~~~~~~~~~~~~~~~
                                ADC1_CHANNEL_7
*** [.pio/build/esp32dev/src/VoltageMeter.cpp.o] Error 1
In file included from src/globals.h:38,
                 from src/bluetooth/_IBluetoothService.h:32,
                 from src/bluetooth/BatteryService.h:27,
                 from src/bluetooth/BatteryService.cpp:24:
src/VoltageMeter.h:34:32: error: 'ADC1_GPIO34_CHANNEL' was not declared in this scope
       adc1_channel_t channel = ADC1_GPIO34_CHANNEL);
                                ^~~~~~~~~~~~~~~~~~~
src/VoltageMeter.h:34:32: note: suggested alternative: 'ADC1_CHANNEL_7'
       adc1_channel_t channel = ADC1_GPIO34_CHANNEL);
                                ^~~~~~~~~~~~~~~~~~~
                                ADC1_CHANNEL_7
*** [.pio/build/esp32dev/src/bluetooth/BatteryService.cpp.o] Error 1
============================================= [FAILED] Took 10.87 seconds =============================================
The terminal process "platformio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

amandel, 2022-05-07:
Thanks for the report. Might be related to recent updates, for me and at least for the last GitHub actions build all was fine. I'll check.

amandel, 2022-05-07:
I can confirm this is caused by the new lib versions.

closed 2022-05-07 by amandel

amandel, 2022-05-07:
At least the compile errors are resolved now. @tfar, thanks again for reporting!

commit c6e7e3b5 (2022-05-07): Changes needed due to ESP_IDF update fixes #305