openbikesensor/OpenBikeSensorFirmware #309: GPS issues

tfar, 2022-05-07:
Hi @amandel

As Discourse running on https://forum.openbikesensor.org has an unfortunate message limit for new users. I'll continue on the issue here.

Here is the log output with the suggested changes as per https://forum.openbikesensor.org/t/keine-kommunikation-mit-gps-modulen-bei-6-sensoren/847/8 .

```
================================================ [SUCCESS] Took 43.13 seconds ================================================
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Miniterm on /dev/cu.usbserial-0001  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
[    40][I][OpenBikeSensorFirmware.cpp:197] setup(): openbikesensor.org - OBS/v0.15local
[    42][W][Wire.cpp:204] begin(): Bus already started in Master Mode.
[    86][I][VoltageMeter.cpp:40] VoltageMeter(): Initializing VoltageMeter.
[    87][I][VoltageMeter.cpp:54] VoltageMeter(): Characterized using eFuse Vref
[    89][I][VoltageMeter.cpp:62] VoltageMeter(): eFuse Two Point: NOT supported
[    96][I][VoltageMeter.cpp:66] VoltageMeter(): eFuse Vref: Supported
[   106][I][VoltageMeter.cpp:75] VoltageMeter(): VoltageMeter initialized got 0.21V.
[   141][W][sd_diskio.cpp:174] sdCommand(): no token received
[   247][W][sd_diskio.cpp:174] sdCommand(): no token received
[   351][I][OpenBikeSensorFirmware.cpp:590] loadConfig(): Load cfg
[  1149][E][HardwareSerial.cpp:278] setRxBufferSize(): RX Buffer can't be resized when Serial is already running.

[  1160][W][gps.cpp:377] handle(): GPS in: 0xb5
[  1160][W][gps.cpp:377] handle(): GPS in: 0x62
[  1161][W][gps.cpp:377] handle(): GPS in: 0x05
[  1162][W][gps.cpp:377] handle(): GPS in: 0x00
[  1166][W][gps.cpp:377] handle(): GPS in: 0x02
[  1170][W][gps.cpp:377] handle(): GPS in: 0x00
[  1175][W][gps.cpp:377] handle(): GPS in: 0x06
[  1179][W][gps.cpp:377] handle(): GPS in: 0x34
[  1183][W][gps.cpp:377] handle(): GPS in: 0x41
[  1187][W][gps.cpp:377] handle(): GPS in: 0x66
[  1192][E][gps.cpp:811] parseUbxMessage(): ACK-NAK 0x3406
[  1197][E][gps.cpp:350] sendAndWaitForAck(): Failed to send cfg. 0x3406 NAK: 1 
[  1241][W][gps.cpp:377] handle(): GPS in: 0xb5
[  1241][W][gps.cpp:377] handle(): GPS in: 0x62
[  1241][W][gps.cpp:377] handle(): GPS in: 0x05
[  1243][W][gps.cpp:377] handle(): GPS in: 0x00
[  1247][W][gps.cpp:377] handle(): GPS in: 0x02
[  1251][W][gps.cpp:377] handle(): GPS in: 0x00
[  1255][W][gps.cpp:377] handle(): GPS in: 0x06
[  1260][W][gps.cpp:377] handle(): GPS in: 0x34
[  1264][W][gps.cpp:377] handle(): GPS in: 0x41
[  1268][W][gps.cpp:377] handle(): GPS in: 0x66
[  1272][E][gps.cpp:811] parseUbxMessage(): ACK-NAK 0x3406
[  1278][E][gps.cpp:350] sendAndWaitForAck(): Failed to send cfg. 0x3406 NAK: 1 
[  1285][E][gps.cpp:293] setBaud(): Switch to 115200 was not possible, back to 9600.
[  2292][E][gps.cpp:345] sendAndWaitForAck(): Retry to send 0x3406
[  3292][E][gps.cpp:345] sendAndWaitForAck(): Retry to send 0x3406
[  4292][E][gps.cpp:345] sendAndWaitForAck(): Retry to send 0x3406
[  4292][E][gps.cpp:350] sendAndWaitForAck(): Failed to send cfg. 0x3406 NAK: 0 
[  4294][E][gps.cpp:298] setBaud(): NO GPS????
[ 10298][E][gps.cpp:345] sendAndWaitForAck(): Retry to send 0x3406
[ 11298][E][gps.cpp:345] sendAndWaitForAck(): Retry to send 0x3406
[ 12298][E][gps.cpp:345] sendAndWaitForAck(): Retry to send 0x3406
[ 12298][E][gps.cpp:350] sendAndWaitForAck(): Failed to send cfg. 0x3406 NAK: 0 
[ 13300][E][gps.cpp:345] sendAndWaitForAck(): Retry to send 0x3406
[ 14300][E][gps.cpp:345] sendAndWaitForAck(): Retry to send 0x3406
[ 15300][E][gps.cpp:345] sendAndWaitForAck(): Retry to send 0x3406
[ 15300][E][gps.cpp:350] sendAndWaitForAck(): Failed to send cfg. 0x3406 NAK: 0 
[ 15302][I][gps.cpp:174] softResetGps(): Soft-RESET GPS!
[ 16307][E][gps.cpp:345] sendAndWaitForAck(): Retry to send 0x0406
[ 17307][E][gps.cpp:345] sendAndWaitForAck(): Retry to send 0x0406
[ 18307][E][gps.cpp:345] sendAndWaitForAck(): Retry to send 0x0406
[ 18307][E][gps.cpp:350] sendAndWaitForAck(): Failed to send cfg. 0x0406 NAK: 0 
[ 18510][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 18614][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 18715][I][gps.cpp:207] enableAlpIfDataIsAvailable(): Enable ALP
[ 19715][E][gps.cpp:345] sendAndWaitForAck(): Retry to send 0x0106
[ 20715][E][gps.cpp:345] sendAndWaitForAck(): Retry to send 0x0106
[ 21715][E][gps.cpp:345] sendAndWaitForAck(): Retry to send 0x0106
[ 21715][E][gps.cpp:350] sendAndWaitForAck(): Failed to send cfg. 0x0106 NAK: 0 

```

amandel, 2022-05-07:
That was helpful. Can you please test with the latest from the master branch? Change: https://github.com/openbikesensor/OpenBikeSensorFirmware/commit/e8a395e2d1eaf36d6fb9e9a642bb7a8368dc18a8

tfar, 2022-05-07:
With latest master, and your logging patch I get the following logs:
```
[    40][I][OpenBikeSensorFirmware.cpp:197] setup(): openbikesensor.org - OBS/v0.15local
[    42][W][Wire.cpp:204] begin(): Bus already started in Master Mode.
[    86][I][VoltageMeter.cpp:40] VoltageMeter(): Initializing VoltageMeter.
[    87][I][VoltageMeter.cpp:54] VoltageMeter(): Characterized using eFuse Vref
[    89][I][VoltageMeter.cpp:62] VoltageMeter(): eFuse Two Point: NOT supported
[    96][I][VoltageMeter.cpp:66] VoltageMeter(): eFuse Vref: Supported
[   106][I][VoltageMeter.cpp:75] VoltageMeter(): VoltageMeter initialized got 4.02V.
[   480][W][sd_diskio.cpp:174] sdCommand(): no token received
[   586][W][sd_diskio.cpp:174] sdCommand(): no token received
[   690][I][OpenBikeSensorFirmware.cpp:590] loadConfig(): Load cfg
{
  "obs": [
    {
      "obsName": "OpenBikeSensor-30e0",
      "name": "default",
      "simRa": false,
      "bluetooth": false,
      "offset": [
        35,
        35
      ],
      "portalUrl": "https://portal.openbikesensor.org",
      "httpPin": "302201",
      "portalToken": "",
      "displayConfig": 525,
      "privacyConfig": 1,
      "confirmationTimeSeconds": 5,
      "devConfig": 0,
      "selectedPreset": 0,
      "privacyArea": []
    }
  ]
}Dumping current configuration, current selected profile is 0:
{
  "obs": [
    {
      "obsName": "OpenBikeSensor-30e0",
      "name": "default",
      "simRa": false,
      "bluetooth": false,
      "offset": [
        35,
        35
      ],
      "portalUrl": "https://portal.openbikesensor.org",
      "httpPin": "302201",
      "portalToken": "",
      "displayConfig": 525,
      "privacyConfig": 1,
      "confirmationTimeSeconds": 5,
      "devConfig": 0,
      "selectedPreset": 0,
      "privacyArea": []
    }
  ]
}
[  2178][W][gps.cpp:377] handle(): GPS in: 0x00
[  2179][W][gps.cpp:377] handle(): GPS in: 0x00
[  2179][W][gps.cpp:377] handle(): GPS in: 0x00
[  2180][W][gps.cpp:377] handle(): GPS in: 0x00
[  2184][W][gps.cpp:377] handle(): GPS in: 0x00
[  2189][W][gps.cpp:377] handle(): GPS in: 0x00
[  2193][W][gps.cpp:377] handle(): GPS in: 0x00
[  2197][W][gps.cpp:377] handle(): GPS in: 0x00
[  2201][W][gps.cpp:377] handle(): GPS in: 0x00
[  2206][W][gps.cpp:377] handle(): GPS in: 0x00
[  2210][W][gps.cpp:377] handle(): GPS in: 0x00
[  2214][W][gps.cpp:377] handle(): GPS in: 0x00
[  2218][W][gps.cpp:377] handle(): GPS in: 0x00
[  2223][W][gps.cpp:377] handle(): GPS in: 0x00
[  2227][W][gps.cpp:377] handle(): GPS in: 0x00
[  2231][W][gps.cpp:377] handle(): GPS in: 0x00
[  2235][W][gps.cpp:377] handle(): GPS in: 0x00
… this goes on forever –
```

amandel, 2022-05-07:
🤔 🤡 can you please remove the ESP && GPS module from power for some seconds and try again. If this does not change something, please also test without the additional logging.

tfar, 2022-05-07:
I've removed the logging patch. This looks like progress:
```
================================================ [SUCCESS] Took 77.69 seconds ================================================
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Miniterm on /dev/cu.usbserial-0001  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
[    40][I][OpenBikeSensorFirmware.cpp:197] setup(): openbikesensor.org - OBS/v0.15local
[    42][W][Wire.cpp:204] begin(): Bus already started in Master Mode.
[    86][I][VoltageMeter.cpp:40] VoltageMeter(): Initializing VoltageMeter.
[    87][I][VoltageMeter.cpp:54] VoltageMeter(): Characterized using eFuse Vref
[    89][I][VoltageMeter.cpp:62] VoltageMeter(): eFuse Two Point: NOT supported
[    96][I][VoltageMeter.cpp:66] VoltageMeter(): eFuse Vref: Supported
[   106][I][VoltageMeter.cpp:75] VoltageMeter(): VoltageMeter initialized got 0.21V.
[   141][W][sd_diskio.cpp:174] sdCommand(): no token received
[   247][W][sd_diskio.cpp:174] sdCommand(): no token received
[   351][I][OpenBikeSensorFirmware.cpp:590] loadConfig(): Load cfg
{
  "obs": [
    {
      "obsName": "OpenBikeSensor-30e0",
      "name": "default",
      "simRa": false,
      "bluetooth": false,
      "offset": [
        35,
        35
      ],
      "portalUrl": "https://portal.openbikesensor.org",
      "httpPin": "302201",
      "portalToken": "",
      "displayConfig": 525,
      "privacyConfig": 1,
      "confirmationTimeSeconds": 5,
      "devConfig": 0,
      "selectedPreset": 0,
      "privacyArea": []
    }
  ]
}Dumping current configuration, current selected profile is 0:
{
  "obs": [
    {
      "obsName": "OpenBikeSensor-30e0",
      "name": "default",
      "simRa": false,
      "bluetooth": false,
      "offset": [
        35,
        35
      ],
      "portalUrl": "https://portal.openbikesensor.org",
      "httpPin": "302201",
      "portalToken": "",
      "displayConfig": 525,
      "privacyConfig": 1,
      "confirmationTimeSeconds": 5,
      "devConfig": 0,
      "selectedPreset": 0,
      "privacyArea": []
    }
  ]
}
[  2149][E][gps.cpp:345] sendAndWaitForAck(): Retry to send 0x3406
[  3150][E][gps.cpp:345] sendAndWaitForAck(): Retry to send 0x3406
[  4150][E][gps.cpp:345] sendAndWaitForAck(): Retry to send 0x3406
[  4150][E][gps.cpp:350] sendAndWaitForAck(): Failed to send cfg. 0x3406 NAK: 0 
[  5181][E][gps.cpp:345] sendAndWaitForAck(): Retry to send 0x3406
[  5192][E][gps.cpp:811] parseUbxMessage(): ACK-NAK 0x3406
[  5193][E][gps.cpp:350] sendAndWaitForAck(): Failed to send cfg. 0x3406 NAK: 1 
[... 102 lines cut ...]
[ 25621][I][gps.cpp:1056] handleUbxNavTimeGps(): TIMEGPS: iTOW: 171000, fTOW: 0, week 0, leapS: 14, valid: 0x00 (), tAcc 4294967295ns, DATE: 1980-01-06T00:02:51, delay 3ms
[ 26621][I][gps.cpp:1056] handleUbxNavTimeGps(): TIMEGPS: iTOW: 172000, fTOW: 0, week 0, leapS: 14, valid: 0x00 (), tAcc 4294967295ns, DATE: 1980-01-06T00:02:52, delay 3ms
[ 27621][I][gps.cpp:1056] handleUbxNavTimeGps(): TIMEGPS: iTOW: 173000, fTOW: 0, week 0, leapS: 14, valid: 0x00 (), tAcc 4294967295ns, DATE: 1980-01-06T00:02:53, delay 3ms
[ 28621][I][gps.cpp:1056] handleUbxNavTimeGps(): TIMEGPS: iTOW: 174000, fTOW: 0, week 0, leapS: 14, valid: 0x00 (), tAcc 4294967295ns, DATE: 1980-01-06T00:02:54, delay 3ms
[ 29621][I][gps.cpp:1056] handleUbxNavTimeGps(): TIMEGPS: iTOW: 175000, fTOW: 0, week 0, leapS: 14, valid: 0x00 (), tAcc 4294967295ns, DATE: 1980-01-06T00:02:55, delay 3ms
[ 30621][I][gps.cpp:1056] handleUbxNavTimeGps(): TIMEGPS: iTOW: 176000, fTOW: 0, week 0, leapS: 14, valid: 0x00 (), tAcc 4294967295ns, DATE: 1980-01-06T00:02:56, delay 3ms
[ 31621][I][gps.cpp:1056] handleUbxNavTimeGps(): TIMEGPS: iTOW: 177000, fTOW: 0, week 0, leapS: 14, valid: 0x00 (), tAcc 4294967295ns, DATE: 1980-01-06T00:02:57, delay 3ms
[ 32621][I][gps.cpp:1056] handleUbxNavTimeGps(): TIMEGPS: iTOW: 178000, fTOW: 0, week 0, leapS: 14, valid: 0x00 (), tAcc 4294967295ns, DATE: 1980-01-06T00:02:58, delay 3ms
[ 33621][I][gps.cpp:1056] handleUbxNavTimeGps(): TIMEGPS: iTOW: 179000, fTOW: 0, week 0, leapS: 14, valid: 0x00 (), tAcc 4294967295ns, DATE: 1980-01-06T00:02:59, delay 3ms
[ 34621][I][gps.cpp:1056] handleUbxNavTimeGps(): TIMEGPS: iTOW: 180000, fTOW: 0, week 0, leapS: 14, valid: 0x00 (), tAcc 4294967295ns, DATE: 1980-01-06T00:03:00, delay 3ms
[ 35621][I][gps.cpp:1056] handleUbxNavTimeGps(): TIMEGPS: iTOW: 181000, fTOW: 0, week 0, leapS: 14, valid: 0x00 (), tAcc 4294967295ns, DATE: 1980-01-06T00:03:01, delay 3ms
[ 36621][I][gps.cpp:1056] handleUbxNavTimeGps(): TIMEGPS: iTOW: 182000, fTOW: 0, week 0, leapS: 14, valid: 0x00 (), tAcc 4294967295ns, DATE: 1980-01-06T00:03:02, delay 3ms
[ 37621][I][gps.cpp:1056] handleUbxNavTimeGps(): TIMEGPS: iTOW: 183000, fTOW: 0, week 0, leapS: 14, valid: 0x00 (), tAcc 4294967295ns, DATE: 1980-01-06T00:03:03, delay 3ms
[ 38622][I][gps.cpp:1056] handleUbxNavTimeGps(): TIMEGPS: iTOW: 184000, fTOW: 0, week 0, leapS: 14, valid: 0x00 (), tAcc 4294967295ns, DATE: 1980-01-06T00:03:04, delay 3ms
[ 39622][I][gps.cpp:1056] handleUbxNavTimeGps(): TIMEGPS: iTOW: 185000, fTOW: 0, week 0, leapS: 14, valid: 0x00 (), tAcc 4294967295ns, DATE: 1980-01-06T00:03:05, delay 3ms
[... cut ...]
[ 46622][I][gps.cpp:1056] handleUbxNavTimeGps(): TIMEGPS: iTOW: 192000, fTOW: 0, week 0, leapS: 14, valid: 0x00 (), tAcc 4294967295ns, DATE: 1980-01-06T00:03:12, delay 4ms
[ 47622][I][gps.cpp:1056] handleUbxNavTimeGps(): TIMEGPS: iTOW: 193000, fTOW: 0, week 0, leapS: 14, valid: 0x00 (), tAcc 4294967295ns, DATE: 1980-01-06T00:03:13, delay 3ms
[ 48622][I][gps.cpp:1056] handleUbxNavTimeGps(): TIMEGPS: iTOW: 194000, fTOW: 0, week 0, leapS: 14, valid: 0x00 (), tAcc 4294967295ns, DATE: 1980-01-06T00:03:14, delay 4ms
```

amandel, 2022-05-07:
Way better! Is waiting for GPS Time and Fix and now. I'll check for the other messages, looks like this is a different version of the firmware.

amandel, 2022-05-07:
On more test to make it more stable - can you please change line https://github.com/openbikesensor/OpenBikeSensorFirmware/blob/master/src/gps.cpp#L321 to `const int timeoutMs = 3000;`

tfar, 2022-05-07:
It is probably unrelated to the timeout change, but now i get the following new lines about unexpected GPS char:
```
…
  5200][I][gps.cpp:207] enableAlpIfDataIsAvailable(): Enable ALP
[  5204][I][gps.cpp:1055] handleUbxNavTimeGps(): TIMEGPS: iTOW: 1066000, fTOW: 0, week 0, leapS: 14, valid: 0x00 (), tAcc 4294967295ns, DATE: 1980-01-06T00:17:46, delay 3895ms
[  5336][I][gps.cpp:419] addStatisticsMessage(): New: ALP Data valid from: 2022-05-01T13:00:00
[  5337][I][gps.cpp:419] addStatisticsMessage(): New: ALP Data valid to: 2022-05-15T13:00:00
[  5359][E][gps.cpp:810] parseUbxMessage(): ACK-NAK 0x3406
[  5360][E][gps.cpp:350] sendAndWaitForAck(): Failed to send cfg. 0x3406 NAK: 1 
[  5410][W][gps.cpp:632] encode(): Unexpected GPS char in state null: 8a �
[  5410][W][gps.cpp:632] encode(): Unexpected GPS char in state null: ff �
[  5412][W][gps.cpp:632] encode(): Unexpected GPS char in state null: 01 ␁
[  5419][W][gps.cpp:632] encode(): Unexpected GPS char in state null: 58 X
[  5425][W][gps.cpp:632] encode(): Unexpected GPS char in state null: 6b k
[  5434][I][gps.cpp:419] addStatisticsMessage(): New: swVersion: 5.00 (28483)
[  5439][I][gps.cpp:419] addStatisticsMessage(): New: hwVersion: 00040005
…
```

amandel, 2022-05-07:
Unexpected data can be caused by random communication error or by full buffers. For the full buffer you should also get a additional `readGPSData` message. Is the data always the same after a reset?

But looks like the increased timeout did help (a bit). ✅

tfar, 2022-05-07:
> Is the data always the same after a reset?

What do you mean by reset exactly? 

1. The `EN` button which seems to cause a restart of the firmware.
2. Removing battery and USB power and reconnecting the power.

BTW: Does the setup need both battery and USB power or should USB be sufficient for the whole board? With USB only the display and all other stuff at least seems to work.

amandel, 2022-05-07:
Pushing `EN` is fine in this case. If the same data comes in every time it is likely not a temporary communication error and worth to look deeper at it.

USB only can be to little for all parts to work reliably, but I can not answer this 100%.

Can you share a full log with the last changes?

tfar, 2022-05-07:
This is the full log after pressing `EN` button.

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12776
load:0x40080400,len:3032
entry 0x400805e4
[    40][I][OpenBikeSensorFirmware.cpp:197] setup(): openbikesensor.org - OBS/v0.15local
[    42][W][Wire.cpp:204] begin(): Bus already started in Master Mode.
[    86][I][VoltageMeter.cpp:40] VoltageMeter(): Initializing VoltageMeter.
[    87][I][VoltageMeter.cpp:54] VoltageMeter(): Characterized using eFuse Vref
[    89][I][VoltageMeter.cpp:62] VoltageMeter(): eFuse Two Point: NOT supported
[    96][I][VoltageMeter.cpp:66] VoltageMeter(): eFuse Vref: Supported
[   106][I][VoltageMeter.cpp:75] VoltageMeter(): VoltageMeter initialized got 4.02V.
[   479][W][sd_diskio.cpp:174] sdCommand(): no token received
[   585][W][sd_diskio.cpp:174] sdCommand(): no token received
[   689][I][OpenBikeSensorFirmware.cpp:590] loadConfig(): Load cfg
{
  "obs": [
    {
      "obsName": "OpenBikeSensor-30e0",
      "name": "default",
      "simRa": false,
      "bluetooth": false,
      "offset": [
        35,
        35
      ],
      "portalUrl": "https://portal.openbikesensor.org",
      "httpPin": "302201",
      "portalToken": "",
      "displayConfig": 525,
      "privacyConfig": 1,
      "confirmationTimeSeconds": 5,
      "devConfig": 0,
      "selectedPreset": 0,
      "privacyArea": []
    }
  ]
}Dumping current configuration, current selected profile is 0:
{
  "obs": [
    {
      "obsName": "OpenBikeSensor-30e0",
      "name": "default",
      "simRa": false,
      "bluetooth": false,
      "offset": [
        35,
        35
      ],
      "portalUrl": "https://portal.openbikesensor.org",
      "httpPin": "302201",
      "portalToken": "",
      "displayConfig": 525,
      "privacyConfig": 1,
      "confirmationTimeSeconds": 5,
      "devConfig": 0,
      "selectedPreset": 0,
      "privacyArea": []
    }
  ]
}
[  1497][E][gps.cpp:810] parseUbxMessage(): ACK-NAK 0x3406
[  1498][E][gps.cpp:350] sendAndWaitForAck(): Failed to send cfg. 0x3406 NAK: 1 
[  1499][I][gps.cpp:275] setBaud(): GPS startup already 115200
[... 104 lines cut ...]
[  7923][I][gps.cpp:1055] handleUbxNavTimeGps(): TIMEGPS: iTOW: 17000, fTOW: 0, week 0, leapS: 14, valid: 0x00 (), tAcc 4294967295ns, DATE: 1980-01-06T00:00:17, delay 3ms
[  8468][E][gps.cpp:345] sendAndWaitForAck(): Retry to send 0x0106
[  8595][I][gps.cpp:419] addStatisticsMessage(): New: ALP Data valid from: 2022-05-01T13:00:00
[  8596][I][gps.cpp:419] addStatisticsMessage(): New: ALP Data valid to: 2022-05-15T13:00:00
[  8617][E][gps.cpp:810] parseUbxMessage(): ACK-NAK 0x3406
[  8618][E][gps.cpp:350] sendAndWaitForAck(): Failed to send cfg. 0x3406 NAK: 1 
[  8677][W][sd_diskio.cpp:174] sdCommand(): no token received
[  8782][W][sd_diskio.cpp:174] sdCommand(): no token received
[  8886][W][sd_diskio.cpp:174] sdCommand(): no token received
[  8990][W][sd_diskio.cpp:174] sdCommand(): no token received
[  9100][W][sd_diskio.cpp:174] sdCommand(): no token received
[  9203][W][sd_diskio.cpp:174] sdCommand(): no token received
[  9313][W][sd_diskio.cpp:174] sdCommand(): no token received
[  9419][I][gps.cpp:419] addStatisticsMessage(): New: readGPSData(av: 338 bytes in buffer, lastCall 751ms ago, at 1970-01-01T00:00:09)
[  9424][I][gps.cpp:419] addStatisticsMessage(): New: swVersion: 5.00 (28483)
[  9427][I][gps.cpp:419] addStatisticsMessage(): New: hwVersion: 00040005
[  9440][I][gps.cpp:1055] handleUbxNavTimeGps(): TIMEGPS: iTOW: 18000, fTOW: 0, week 0, leapS: 14, valid: 0x00 (), tAcc 4294967295ns, DATE: 1980-01-06T00:00:18, delay 773ms
[  9448][W][Wire.cpp:204] begin(): Bus already started in Master Mode.
[  9923][I][gps.cpp:1055] handleUbxNavTimeGps(): TIMEGPS: iTOW: 19000, fTOW: 0, week 0, leapS: 14, valid: 0x00 (), tAcc 4294967295ns, DATE: 1980-01-06T00:00:19, delay 2ms
[ 10923][I][gps.cpp:1055] handleUbxNavTimeGps(): TIMEGPS: iTOW: 20000, fTOW: 0, week 0, leapS: 14, valid: 0x00 (), tAcc 4294967295ns, DATE: 1980-01-06T00:00:20, delay 2ms
[ 11923][I][gps.cpp:1055] handleUbxNavTimeGps(): TIMEGPS: iTOW: 21000, fTOW: 0, week 0, leapS: 14, valid: 0x00 (), tAcc 4294967295ns, DATE: 1980-01-06T00:00:21, delay 2ms
[ 12923][I][gps.cpp:1055] handleUbxNavTimeGps(): TIMEGPS: iTOW: 22000, fTOW: 0, week 0, leapS: 14, valid: 0x00 (), tAcc 4294967295ns, DATE: 1980-01-06T00:00:22, delay 2ms
[ 13923][I][gps.cpp:1055] handleUbxNavTimeGps(): TIMEGPS: iTOW: 23000, fTOW: 0, week 0, leapS: 14, valid: 0x00 (), tAcc 4294967295ns, DATE: 1980-01-06T00:00:23, delay 2ms
[ 14923][I][gps.cpp:1055] handleUbxNavTimeGps(): TIMEGPS: 
```

amandel, 2022-05-07:
`setBaud(): GPS startup already 115200` -> Top, configuration was stored

`[  5349][E][gps.cpp:381] handle(): break after 1048 bytes` there was a lot of data from the GPS module, the firmware stopped reading after 1048 bytes. 

`[  5906][W][gps.cpp:672] encode(): UBX CK_A error: ef != 44 after 59 bytes for 0x0601` there was a invalid data read from the GPS module (CRC check failure)
`[  5906][W][gps.cpp:632] encode(): Unexpected GPS char in state null: 80 �` as above some communication error, which might lead to lost GPS messages. 

`[  9419][I][gps.cpp:419] addStatisticsMessage(): New: readGPSData(av: 338 bytes in buffer, lastCall 751ms ago, at 1970-01-01T00:00:09)` a lot of GPS data was received since the firmware read GPS data last time, this is fine during startup. Also since we have a buffer of 512bytes the 338 bytes that piled up in the buffer are still fine.

One one side there are signs for unstable communication, after the startup this seems to stabilize and you receive valid `NAV-TIMEGPS` messages every second, with no garbage or error. 

I hope you get time and then a GPS fix after a while. Your module has an older firmware than most of the modules. Your does report `swVersion: 5.00 (28483)` while what I typically see is `swVersion: 7.03 (45969)`. So it is interesting to see if there are additional gps errors once the gps fix is established.

pReya, 2022-05-08:
@amandel Thank you so much for your debugging help!
@tfar What's the Discourse message limit you're running into? We should get this fixed, so new people aren't blocked from writing helpful stuff on the forum.

tfar, 2022-05-08:
@pReya Maybe my case was an edge case as I posted logs with sensitvie information so I had to delete the old post and repost 🤦‍♂️ without the sensitive information. This took two attempts because I missed some sensitive information in my first repost. So it's either a limit for 3 posts or 3 consecutive posts. Probably also related to my account on there being rather new.
I haven't found a good source on Discourse `limts` and `trust levels`.

tfar, 2022-05-08:
Hi @amandel 

> `[ 5349][E][gps.cpp:381] handle(): break after 1048 bytes` there was a lot of data from the GPS module, the firmware stopped reading after 1048 bytes.
> 
> ``

The ESP32 stopped reading or the firmware of the GPS module stopped reading?

> [ 5906][W][gps.cpp:672] encode(): UBX CK_A error: ef != 44 after 59 bytes for 0x0601 there was a invalid data read from the GPS module (CRC check failure)
> [ 5906][W][gps.cpp:632] encode(): [...]

Will the GPS fireware automatically resend corrupt messages? Can the ESP32 request the corrupt messages again?

> I hope you get time and then a GPS fix after a while. Your module has an older firmware than most of the modules. Your does report swVersion: 5.00 (28483) while what I typically see is swVersion: [...]

So far no success. Even after letting it run for more than 5 hours. Any other options I can try out? Would it be worth trying to connect the GPS module with a UART to USB adapter and updating its firmware, if possible?

amandel, 2022-05-08:
> The ESP32 stopped reading or the firmware of the GPS module stopped reading?

It "paused" the reading to give other tasks the opportunity to do the work. Reading data continues after this.

> Will the GPS firmware automatically resend corrupt messages? Can the ESP32 request the corrupt messages again?

This depends. Configuration messages from ESP->GPS are sent 3 times if no ack is received within the now 3 second timeout. Most of the messages GPS->ESP are sent in a certain interval we will lose one update if there is a transfer issue. But catch up with the next. So this is nothing critical if it does not happen to often.

> So far no success.

😞 - did the content of the `TIMEGPS` message change at all? Does the antenna have a good chance to get the signal? I assume you know the [[Wiki] GPS-Troubleshooting / Fehlersuche](https://forum.openbikesensor.org/t/wiki-gps-troubleshooting-fehlersuche/170) already? As far as I know there is no way to update the firmware of the GPS module.

tfar, 2022-05-12:
> [  1498][E][gps.cpp:350] sendAndWaitForAck(): Failed to send cfg. 0x3406 NAK: 1 

@amandel Is this critical? I have the feeling my module does not download the ALP data and/or does not send it to the GPS module?

> [  1985][E][vfs_api.cpp:102] open(): /sd/aid_ini.ubx does not exist, no permits for creation

Does this indicate it tried to download ALP data but failed to write it to the SD card?

amandel, 2022-05-14:
0x3406 is CFG-RINV, this is used by the OBS firmware to detect if the config settings are current or need to be updated. It is supported starting with U-Blox firmware 6.00, so it is not unexpected to fail here also it is not critical for GPS reception. 
`aid_ini.ubx` contains data of the last GPS fix - so if there was no GPS fix ever, it is expected to be missing.
For ALP data please have a look at the [Forum](https://forum.openbikesensor.org/t/wiki-gps-troubleshooting-fehlersuche/170/10?u=andreas) where I tried to describe the details.

Good to hear that it works now. If you have further GPS related errors in the serial log feel free to share. When talking about firmware versions it is always useful to share the real number. There was a ALP fix a while ago, the current [v0.14.731](https://github.com/openbikesensor/OpenBikeSensorFirmware/releases/tag/v0.14.731) does not have a known issue with ALP.

amandel, 2022-08-02:
Fixed :)

closed 2022-08-02 by amandel