openbikesensor/OpenBikeSensorFirmware #326: Measurements are shifted in time and location

macronom, 2022-12-21:
When visualising the measurements saved in the CSV file, I noticed that the measurement results do not match the times which are also saved. They seem to be about three seconds "too early" in the data set. They thus do not fit in terms of location either, as time and location seem to be correctly linked. As a result, the measurements are displaced on the map in the portal. They appear at locations _ahead_ of the actual measurement.

I have been able to reproduce this behaviour on a dozen recordings and with two different devices. I have created a video to illustrate this behaviour. In the video, the measurements are synchronised in time with the recording. You can see the effect on both the left and the right sensor. This applies to all recorded measurements, even those that have not been confirmed.

https://user-images.githubusercontent.com/12209721/208958488-2a9228c6-6a01-49e9-a43f-d5104f5c8cc3.mp4

amandel, 2022-12-22:
Thanks for the detailed report. We had a similar report before that pointed to the confirmed measurements but I could not find a reason for this. But with your input, the error can also be in a different area. I'm on it!

macronom, 2022-12-22:
Thanks @amandel for the reply.

I am aware that there is generally a possible maximum delay of 1 second because the data is only written once per second. In my clip it was pure coincidence that the measurements matched the change in seconds so precisely.

If, as you write, the delay of the GPS data is taken into account, there is a further maximum delay of 1 second in the data.

However, the timer seems unimportant to me in this context. I see that the attempt is made to always write at the beginning of a second. This could only explain the deviation of one second mentioned at the beginning.

So at this point, we can explain a maximum of 2 seconds of deviation, which corresponds to your calculation. On average, it should be one second. However, after watching a handful of synchronised videos, I have the feeling that the average deviation is 2.5 seconds. Well, I know that "feeling" is not a good argument. On the other hand, there seems to be no explanation for the 3 seconds delay in my clip (which indeed seems to be an extreme value).

For my purposes, it would of course be a simple solution to move the measurements "upwards" resp. forward in time, as you suggested. That is what I will do. So in my example video I would add a 3 second delay and measurement and observation would match perfectly, keeping (almost) correct time and location. That is of course just a workaround and not a general solution. My concerns were more about the fact that the location of the measurement is also influenced, which has an impact on the display in the portal. But since it is usually only about whole street sections, this may not be a big problem. 

I see that there is no completely optimal solution to the problem. The fact that the shift always goes in one direction makes it reasonable to build in a compensation. That would at least push the data closer to the true value. The average explainable deviation of one second would be a reasonable choice. However, that would only be a few meters on the map.

amandel, 2022-12-22:
With my changes in #328 we "win" 1-2 seconds:
- I take the GPS position reported *after* the measurements. I see no reason why we must take the value at the beginning or the end. 
- The time is synchronized with the GPS reporting, so the offset of 0-1 sec that could have existed is set to 0.

Now we should be down to 0-1 sec offset which can be compensated by the portal, by interpolating the position between current measurement and next measurement, taking the passed milliseconds into account. 

All is true only if my conclusions are correct :) And - from your explanation we might have an additional gap that is not known until now. But we are heading in the right direction.

@macronom can you see the artifacts at: https://github.com/openbikesensor/OpenBikeSensorFirmware/actions/runs/3758861031 for download? I would be interested what you see with this version.

macronom, 2022-12-22:
I just flashed the test version. I'll try to collect some data tomorrow and maybe make a new video if there's a rain break (I'm not sure if the sensor is as waterproof as I am ;-)

macronom, 2022-12-24:
I did a test drive earlier, but the recorded data looks strange. Here is a confirmed measurement (first line). In the third column are the millis.

24.12.2022 09:26:42 343528 (confirmed measurement)
24.12.2022 09:26:42 344528
24.12.2022 09:26:43 345531
24.12.2022 09:26:44 345533

You can see that exactly 1000 milliseconds have passed between lines 1 and 2, i.e. exactly one second. But the times in column 2 are identical.
From line 3 to line 4 there is only a difference of 2 milliseconds, but the seconds are counted up (which is of course possible, as this could be exactly on one end of the second).

My script for displaying the recorded data works on the basis of the milliseconds, but I need to check first what effect these large fluctuations in the time differences have.

What I am missing is a mapping of the millis to the (exact) time. So for example 343528 millis correspond to 09:26:42.123 (GPS time). This would allow me to ignore the time in column 2 and always calculate the exact time from the millis. Would that be possible, or is this assignment not feasible? I see that in column 4 CPU and GPS time are already sometimes compared, but always only on the basis of whole seconds.

macronom, 2022-12-24:
I have now quickly created a video without making any changes to my script. Surprisingly, you can now see that the overlays are about 2.7 seconds "too late", no longer too early.

You can also see how jumpy the seconds are in the display. This makes it difficult to synchronise the video with the overlay track. Up to one second delay should be explainable by this.

However, since my script is based on milliseconds, the time flow should be correct. The time displayed originates directly from the data and is not calculated using the milliseconds.

I have not yet had time to compare the coordinates of the measurements with the video. However, this should be possible on the basis of the overlaid data.

https://user-images.githubusercontent.com/12209721/209439116-1bce346e-c0d9-4b37-b5ac-63b1cb78460c.mp4

amandel, 2022-12-25:
Thanks a lot for the input! I need to look more deep into this, and likely change back, where we start the loop (with new GPS data). I hope this does not change your good results.

The problem you see now, is likely because the time in the CSV is based on the real time clock (RTC) of the ESP not directly GPS time. Since I now switch based on the timing of the GPS, chances are that we get the time a split second before the tick and next time a split second after the tick. I will change this to use GPS time here at all places. The idea to use RTC here was to still have a stable clock base even if GPS gets week. But GPS clock is way better and keeps reporting (what I see) even if satellite connection is lost.

The millis counter is always linear and reliable. But I see a way to make the value in the CSV more accurate too by aligning it to the time when the GPS message was received. In theory you can sync the time to the millis at any place with good GPS reception.

amandel, 2022-12-27:
I've now changed timing of the measurement-loop so that it fully relies on the GPS time. New GPS time info (TOW) triggers new 1 second of measurement, time and position information is taken from GPS, millis is taken from the moment, when the GPS info arrived. That should be as accurate as it can be. I could not do testing of the code until now and there are corner cases to be covered (no GPS at all, some error cases, unexpected delays). For now there is some additional debug output in the CSV file printing the different available timers in the comments section, this should help if there is strange behavior. 
This assumes that the GPS module gives the position (and time) at the very moment for what it is reported. Until now I did not see any hints that this assumption is wrong.
Likely I do not find the time to go deeper till "next year" ;) Feedback welcome!

macronom, 2023-01-02:
I haven't had time to test RC814 yet, but have done some more test drives with RC810 in the meantime. The locations of all recorded events seemed to be very precise. This seems to solve the main problem.

I have just installed version RC814 and hope to find time for some more tests. I will report on this.

BTW, feature request: Boot into admin mode after a firmware update. I get a bit of a shock every time the device no longer shows up in the Wi-Fi after an update. ;-)

amandel, 2023-01-06:
There is now a new [RC831](https://github.com/openbikesensor/OpenBikeSensorFirmware/actions/runs/3856537357) that fixes a problem with RC814 that could block it from accepting the GPS fix. 

I hope my recent changes do not roll back your improved experience. As written above the reported time changed and now fully relies on the GPS time. So you should not see the same second reported for 2 measurement loops or 2 second jumps.

macronom, 2023-01-06:
I just had time to look at the results from two test drives (with RC 814).

The records look very good. The coordinates of the measurements match almost perfectly with the locations I can see in the video recording. I would estimate that there was hardly more than 5 metres discrepancy, as good as you would just expect with GPS data.

Surprisingly, the recorded times still seem to be ahead of the event in question. One would expect them to be behind at best. However, the difference is less than one second. And in any case, this is only a problem with the visualisation. All that matters is the location of the measurement.

I see in the data that with RC814 the data was written every second again and therefore there are no more "jumping" seconds. This simplifies the visualisation because no additional logic is necessary. Nice.

I have just installed RC831 and will do a test drive tomorrow and possibly make another video or take a few screenshots.

amandel, 2023-01-06:
Nice. The RC831 should not differ in timing to the RC814. I'll check the code if I can find a reason that the time is ahead, for my understanding this should not be the case. A way to check this might be to push the button while there is no distance offered. This should record a overtake event at that point in time when the button was pushed. Can you share your CSV files (might be via mail), you can remove the GPS coordinates if you like, I'm mainly interested in the comments and timing data.

amandel, 2023-01-10:
If there are no negative findings I'd merge this PR and prepare a pre-release with the changes.

commit c2a344bf (2022-12-22): Use GPS timing as main clock to start measurement look

- measurement look is ended if new GPS data is complete
- use new GPS data for collected measurements
- can lead to alternating timing if started without GPS fix
- fixes #326
PR #328 (2023-01-11): Use GPS timing as main clock to start measurement loop
- measurement loop synchronized with GPS
- use GPS data and time for collected measurements
- if measurements are started early, time will be adjusted and completed as available with GPS. 
- without GPS internal millis counter is used.
- fixes #326

Edit: Typo, refresh
commit 53e9ea44 (2023-01-11): Use GPS timing as main clock to start measurement loop (#328)

- measurement loop is ended if new GPS second tick is received
- GPS info received with the last tick is used for the other data collected in the measurement loop
- if started without GPS fix you might see changing wall clock time (no info, then week info, then full time info) 
- only if there is no GPS module, the millis timer of the ESP is used, otherwise the second tick of the GPS module is used. This is available even without any reception.
- GPS serial timing, detect bad timing
- fixes #326
closed 2023-01-11 by amandel