openbikesensor/OpenBikeSensorFirmware #338: Default route is set to debug AP

FabianSchwartau, 2023-03-26:
Hi everyone,
I just noticed, that (I guess) the DHCP server also announces itself as a gateway, so my laptop sets the default route through the wifi interface connected to the OBS, although there is nothing to reach beyond the OBS. As my laptop is connected via Ethernet to the internet, I loose internet connection.
In Linux I can tell the network manager to ignore the routes he got from the AP, but this is just a workaround. Especially when you connect to a lot of devices, this is pretty annoying. Am am not an expert on that stuff, but I am fairly sure that it is possible to announce no gateway/default route from an AP.
Best,
Fabian

amandel, 2023-06-09:
Hi Fabian,
not sure if this is related to the AP or it is a general Windows(?) behavior to treat - by default - the last connected network device also as default gateway. As far as I remember we do not set anything special in regard to DHCP related to a gateway or routes. Might be you or someone found a hint what we need to do to get better results?
Kind regards, Andreas.

amandel, 2026-02-21:
Hi @FabianSchwartau  - can we close this?

FabianSchwartau, 2026-02-21:
Hi,
sorry I have not looked into this any further.

amandel, 2026-02-21:
Thanks for the quick feedback. Lets close it for now and re-open or create a fresh one if needed.