openbikesensor/OpenBikeSensorFirmware #369: OBS hangs at "Wait for GPS" while GPS has fix

Jens-Kassel, 2024-11-27:
- GPS seems to be configured correctly
- attaching a UART-USB Converter to the GPS modul and analysing via pygpsclient shows  valid GPS Data with fix
- OBS hangs in loop and shows „Wait for GPS“

See [Forum](https://forum.openbikesensor.org/t/untersuchungen-an-schwierigen-neo8m-gps-modulen/2326)

amandel, 2024-12-01:
Thanks for the PR. Intention of the check is to avoid to set the time based on an to "old" GPS message. Following the conversation looks like for some cases the 20ms are not sufficient and we better increase but not drop the value altogether.

amandel, 2025-05-16:
Fixed with 0.21.928 - See [Forum](https://forum.openbikesensor.org/t/untersuchungen-an-schwierigen-neo8m-gps-modulen/2326/19?u=andreas )

closed 2025-05-16 by amandel