openbikesensor/OpenBikeSensorFirmware #372: Garbage from PlatformIO Serial Monitor

gdesmott, 2025-01-07:
Most of the time when trying to use the PlatformIO Serial Monitor with my board I get this kind of output:

[image]

Sometimes it does work fine, no idea why. Is there any way to prevent this?

gdesmott, 2025-01-07:
The problem also appears without vscode when starting `platformio device monitor` manually.

gluap, 2025-02-19:
This sounds like an issue with serial communication - mismatched baud rates would be a possible explanation. It sometimes working fine sounds strange though. I have seen this - though very rarely - with flaky usb connections.

I usually run ``pio run -t monitor``

we're serving 115200 baud for monitoring - does ``pio device monitor`` respect the baud rate setting for monitoring from platformio.ini?