openbikesensor/OpenBikeSensorFirmware #373: Unclear which OBS variant to use

gdesmott, 2025-01-07:
As I understand from `platformio.ini`  and `custom_config.ini.example` there are two variants of the hardware: `OBSCLASSIC` and `OBSPRO`.

It's not clear from the doc which one should use. Looking at https://www.openbikesensor.org/docs/ I get there is a "classic" and "lite" version but nothing about the pro one?

jensens, 2025-01-30:
I agree, the firmware selection (difference between the files in the release) is not documented well.

gluap, 2025-02-19:
The list of devices is on the web page under ["Das Gerät"->Gerätevarianten](https://www.openbikesensor.org/device/). I agree this could be more obvious - we should probably also document that a ``-e obs`` is recommended for flashing obs classic using platformio. (my guess is that you built an obs classic - "pro" is still in early beta, lite has a different firmware repo directly linked from its building instructions so it's unlikely that a lite builder would show up here)

commit eebd43d1 (2025-03-05): clarify which devices use this firmware
fixes #373
closed 2025-03-05 by gluap